BatchDrake/SigDigger #135: Linux Mint 20

Sylvain-X86, 2020-12-16:
Appimage run weel on Linux Mint 20 but not start after push button 'Start', with RT2832U RF key

`linux; GNU C++ version 4.8.5 20150623 (Red Hat 4.8.5-16); Boost_105300; UHD_003.010.001.000-0-unknown

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6`

BatchDrake, 2020-12-16:
Hi,

It looks like your RTL-SDR is claimed by the corresponding kernel module. It happens quite often. Unload it first with:

```
$ sudo rmmod dvb_usb_rtl28xxu
```

and try again.

Sylvain-X86, 2020-12-16:
hello, thank you for your work and this fabulous program,

after unloading it, the console returns :

<code>usb_claim_interface error -6
[INFO] Using format CF32.</code>

BatchDrake, 2020-12-16:
You're welcome! 

Interesting. Did `rmmod` succeed? Could you unplug the dongle, plug it back again, run `sudo dmesg | tail` and copy here the output? I suspect that your distro is doing shady stuff like detecting the USB device being plugged and loading the module automatically. If that was the case, we can blacklist it with:

```
$ echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf
```

And repeat: unplug, ensure it was unloaded with `rmmod` and plug it back again.

Sylvain-X86, 2020-12-16:
<code>rmmod</code> have succeed

<code>sudo dmesg | tail</code> return :

<code>[17043.639699] rtl2832_sdr rtl2832_sdr.1.auto: Registered as swradio0
[17043.639701] rtl2832_sdr rtl2832_sdr.1.auto: Realtek RTL2832 SDR attached
[17043.639702] rtl2832_sdr rtl2832_sdr.1.auto: SDR API is still slightly experimental and functionality changes may follow
[17043.651493] Registered IR keymap rc-empty
[17043.651594] rc rc0: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:13.2/usb2/2-3/rc/rc0
[17043.651768] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[17043.651875] input: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:13.2/usb2/2-3/rc/rc0/input22
[17043.653600] usb 2-3: dvb_usb_v2: schedule remote query interval to 200 msecs
[17043.662317] usb 2-3: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
[17043.662403] usbcore: registered new interface driver dvb_usb_rtl28xxu
</code>

BatchDrake, 2020-12-16:
Yep, exactly what I suspected. Your Linux Mint is loading `dvb_usb_rtl28xxu` again. Just blacklist the module with this:

```
$ echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf
```

The thing, these dongles are actually digital TV dongles and Linux detects them as such, so it's not uncommon that some distros react to their presence by loading the corresponding kernel module automatically and preventing userland programs from accessing it directly (it's one of those cases where a GNU/Linux distribution gets too smart). As long as you can block the loading of the module, you should be able to use it as an SDR receiver.

Sylvain-X86, 2020-12-16:
Thank you for your support,

after command

<code>sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf</code>

console does not respond ?

BatchDrake, 2020-12-16:
That's one single command, not two commands. You should write the whole command like this, not breaking it:

```
echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf
```

That's a one-liner command to blacklist the module. What this command does is appending the line `blacklist dvb_usb_rtl28xxu` to the end of the configuration file `/etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf`, creating it in case it didn't exist. You could do it by hand as well, editing the file  `/etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf` as root.

Sylvain-X86, 2020-12-16:
Great, 

after restarting the machine, reception is perfectly active! 

Sincere congratulations on your work and your support, i will now listen to my favourite frequencies :)

Happy Holidays

Muchas gracias

73 de F4APX, South-eastern France

closed 2020-12-16 by BatchDrake