BatchDrake/SigDigger #138: FSK Inspector > Data forwarding Functionality not clear

JulioSDDR, 2021-03-18:
I am trying to stream symbols to do "real-time" decoding.  The layout suggests that the `Data forwarding` would refer to symbols, since it is located after the `Demodulator control` tab. Does not work as expected. At first, I thought it was a format difference but it seems to be outputting totally different data. It would be great to relabel/reorder it so its functionality is more clear

This is a mix of feature requests and GUI change proposal. 
It would be great to be able to stream symbols via TCP/UDP or output them to a UNIX FIFO.

BatchDrake, 2021-03-18:
Got it, I'll give it a look and if it is not way too hard to implement I may include it in 0.2

JulioSDDR, 2021-10-04:
I was reminded of this issue today. 
The signal that currently can be streamed is (as far as i can tell) the signal right before the symbol estimator. I would love to be able to process the final data stream, in order to process the fully demodulated message stream in "real time"