BatchDrake/SigDigger #139: Feature: enhance bookmarks

jedi7, 2021-04-29:
Hi,
it will be nice if the bookmark will store also modulation, bandwidth and possible also other settings. So fast switch to different signal will be possible.

I can try to implement it, when will be time.

BatchDrake, 2021-07-01:
If you manage to code this I can include it in SigDigger 0.2. On the other hand, how do we make this information effective? We just set demod bandwidth and configure the appropriate inspector on the GUI? Or do we actually open the inspector? Maybe one click to set the parameters in the GUI, double click to actually open it?

jedi7, 2021-07-02:
Hi, regarding this I opened this PR: https://github.com/BatchDrake/SuWidgets/pull/13
and https://github.com/BatchDrake/SigDigger/pull/148

jedi7, 2021-07-05:
Changes merged to develop, so closing. Thanks :)

closed 2021-07-05 by jedi7