BatchDrake/SigDigger #144: Error launching appImage for Ubuntu 20.04

suciokhan, 2021-06-11:
```
./SigDigger-full-x86_64.AppImage 
/tmp/.mount_SigDig9lEP7b/usr/bin/SigDigger.app: /tmp/.mount_SigDig9lEP7b/usr/bin/../lib/libm.so.6: version `GLIBC_2.29' not found (required by /lib/x86_64-linux-gnu/libasound.so.2)
/tmp/.mount_SigDig9lEP7b/usr/bin/SigDigger.app: /tmp/.mount_SigDig9lEP7b/usr/bin/../lib/libm.so.6: version `GLIBC_2.29' not found (required by /lib/x86_64-linux-gnu/libpng16.so.16)
/tmp/.mount_SigDig9lEP7b/usr/bin/SigDigger.app: /tmp/.mount_SigDig9lEP7b/usr/bin/../lib/libz.so.1: version `ZLIB_1.2.9' not found (required by /lib/x86_64-linux-gnu/libpng16.so.16)
```

BatchDrake, 2021-06-11:
Hi,

Could you please attach the output of the command `apt show libc6` to the issue?

Thanks,

suciokhan, 2021-06-11:
`apt show libc6 -a`
Package: libc6
Version: 2.31-0ubuntu9.3
Status: install ok installed
Priority: required
Section: libs
Source: glibc
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: GNU Libc Maintainers <debian-glibc@lists.debian.org>
Installed-Size: 13.6 MB
Depends: libgcc-s1, libcrypt1 (>= 1:4.4.10-10ubuntu4)
Recommends: libidn2-0 (>= 2.0.5~)
Suggests: glibc-doc, debconf | debconf-2.0, locales
Conflicts: openrc (<< 0.27-2~)
Breaks: hurd (<< 1:0.9.git20170910-1), iraf-fitsutil (<< 2018.07.06-4), libtirpc1 (<< 0.2.3), locales (<< 2.31), locales-all (<< 2.31), nocache (<< 1.1-1~), nscd (<< 2.31), r-cran-later (<< 0.7.5+dfsg-2), wcc (<< 0.0.2+dfsg-3)
Replaces: libc6-amd64
Homepage: https://www.gnu.org/software/libc/libc.html
Original-Vcs-Browser: https://salsa.debian.org/glibc-team/glibc
Original-Vcs-Git: https://salsa.debian.org/glibc-team/glibc.git
Download-Size: unknown
APT-Manual-Installed: no
APT-Sources: /var/lib/dpkg/status
Description: GNU C Library: Shared libraries
 Contains the standard libraries that are used by nearly all programs on
 the system. This package includes shared versions of the standard C library
 and the standard math library, as well as many others.

Package: libc6
Version: 2.31-0ubuntu9.2
Priority: required
Section: libs
Source: glibc
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: GNU Libc Maintainers <debian-glibc@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 13.6 MB
Depends: libgcc-s1, libcrypt1 (>= 1:4.4.10-10ubuntu4)
Recommends: libidn2-0 (>= 2.0.5~)
Suggests: glibc-doc, debconf | debconf-2.0, locales
Conflicts: openrc (<< 0.27-2~)
Breaks: hurd (<< 1:0.9.git20170910-1), iraf-fitsutil (<< 2018.07.06-4), libtirpc1 (<< 0.2.3), locales (<< 2.31), locales-all (<< 2.31), nocache (<< 1.1-1~), nscd (<< 2.31), r-cran-later (<< 0.7.5+dfsg-2), wcc (<< 0.0.2+dfsg-3)
Replaces: libc6-amd64
Homepage: https://www.gnu.org/software/libc/libc.html
Task: minimal
Original-Vcs-Browser: https://salsa.debian.org/glibc-team/glibc
Original-Vcs-Git: https://salsa.debian.org/glibc-team/glibc.git
Download-Size: 2,715 kB
APT-Sources: http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages
Description: GNU C Library: Shared libraries
 Contains the standard libraries that are used by nearly all programs on
 the system. This package includes shared versions of the standard C library
 and the standard math library, as well as many others.

Package: libc6
Version: 2.31-0ubuntu9
Priority: required
Section: libs
Source: glibc
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: GNU Libc Maintainers <debian-glibc@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 13.6 MB
Depends: libgcc-s1, libcrypt1 (>= 1:4.4.10-10ubuntu4)
Recommends: libidn2-0 (>= 2.0.5~)
Suggests: glibc-doc, debconf | debconf-2.0, locales
Conflicts: openrc (<< 0.27-2~)
Breaks: hurd (<< 1:0.9.git20170910-1), iraf-fitsutil (<< 2018.07.06-4), libtirpc1 (<< 0.2.3), locales (<< 2.31), locales-all (<< 2.31), nocache (<< 1.1-1~), nscd (<< 2.31), r-cran-later (<< 0.7.5+dfsg-2), wcc (<< 0.0.2+dfsg-3)
Replaces: libc6-amd64
Homepage: https://www.gnu.org/software/libc/libc.html
Task: minimal
Download-Size: 2,713 kB
APT-Sources: http://us.archive.ubuntu.com/ubuntu focal/main amd64 Packages
Description: GNU C Library: Shared libraries
 Contains the standard libraries that are used by nearly all programs on
 the system. This package includes shared versions of the standard C library
 and the standard math library, as well as many others.

BatchDrake, 2021-06-11:
I see now, there is a bunch of libraries that should not be inside the AppImage to begin with. This may be part of a bigger issue reported here: https://appimage.github.io/SigDigger/

I'm marking this as a bug to be fixed in 0.2 (which I plan to release this Summer), thanks for reporting!

BatchDrake, 2023-04-12:
Hi,

AppImages have changed a lot since this last issue. Is this still crashing for you?

Cheers,

suciokhan, 2023-04-14:
Apologies for the delay- it indeed opens fine now :+1:

closed 2023-04-14 by suciokhan