BatchDrake/SigDigger #149: Unwanted HistogramDialog shows on click after start

jedi7, 2021-07-07:
Steps to reproduce:

1) start SigDigger
2) click on the waterfall
3) the dialog windows HistogramDialog and Sampling windows shows

Showing of HistogramDialog is triggered by HistogramDialog::blanked ->  TimeWindow::onTriggerHistogram -> show histogram

this code is suspicious (HistogramDialog.cpp: 76):
```
  connect(
        this->ui->histogram,
        SIGNAL(blanked()),
        this,
        SIGNAL(blanked()));
```

Any hints how to debug it more?

BatchDrake, 2021-07-07:
I changed the slot to the blanked signal, to prevent opening a new Windows if it is currently closed, changes in develop. Give it a try.

jedi7, 2021-07-08:
yeah that's worked HistogramDialog is gone. 
But Sampling window remains still showing. Maybe the same change will help?

BatchDrake, 2021-07-08:
Done, give it a try

jedi7, 2021-07-08:
great , working :) thanks

closed 2021-07-08 by jedi7