BatchDrake/SigDigger #153: SDRplay RSPdx not appearing as capture device

amerkert, 2021-07-19:
My SDRplay RSPdx is not appearing as capture device, using both SigDigger-0.1.0-x86_64-full.AppImage and/or SigDigger-0.2.0-x86_64-full.AppImage

Linux Mint 20.2/5.4.0-77-generic

All installations of SDRplay API, SoapySDR, and SoapySDRPlay3 were successful.

Install dependencies...

sudo apt install cmake g++ git swig

Install SDRplay API...

Download API/Driver from SDRplay,

sudo ./SDRplay_RSP_API-Linux-3.07.1.run

Install SoapySDR...

git clone https://github.com/pothosware/SoapySDR
cd SoapySDR
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

Install SoapySDRplay3...

git clone https://github.com/pothosware/SoapySDRPlay3
cd SoapySDRPlay3
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

Test if SDRplay device is present...

SoapySDRUtil --probe="driver=sdrplay"

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
[INFO] devIdx: 0
[INFO] hwVer: 4
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSPdx
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=4

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Antenna C
  Corrections: DC removal
  Full gain range: [0, 66] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 27] dB
  Full freq range: [0.001, 2000] MHz
    RF freq range: [0.001, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.096, 0.125, 0.192, 0.25, ..., 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

SoapySDRUtil --info

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g265c3435
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-206b241)
Available factories... sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

amerkert, 2021-07-19:
> IMHO you can try workaround with soapyremote.

Ah! Good temporary workaround! That should work. I would still, of course, like to work on a solution to this issue, so it will work natively.

Thanks!

amerkert, 2021-07-20:
I do have a second RSPdx, everything is working the same on that one, so it is definitely not a hardware issue with that particular SDR. On that note, I am willing to loan my second RSPdx (with paid return postage included), if it will help with this issue, and also if it will improve support for this device.

BatchDrake, 2021-07-20:
Hi @amerkert, and thanks for the offering! I think there's still a lot of room for experimentation on my side before delivering stuff by postal mail / express delivery. In particular, I was suspicious of the SDRPlay driver included in this release (0.2.0) because, to begin with, it was a precompiled binary. Also, I believe I built an older version of the SoapySDR wrapper.

So what I will do these days is to repeat the build with an up-to-date version of the drivers and upload an engineering AppImage for you to test.

PS: do you have any computer you can leave unattended with the SDRPlay plugged in, so I can ssh to it and test things on my own? That could speed up things a little.

amerkert, 2021-07-20:
> Still, I think there's a lot of room for experimentation on my side before

Greetins! I can absolutely set up a machine that you can ssh into, I
appreciate the help, I will set that up today and get back to you with the
ip address and login credentials.

Thanks!

Adam Merkert

closed 2023-04-12 by BatchDrake