BatchDrake/SigDigger #155: RC1 0.2 appImage not starting on Ubuntu 1804 in OracleVirtualBox

waterwin, 2021-08-09:
SigDigger-0.2.0-x86_64-full.AppImage

erwin@UBU2-VirtualBox:~/Downloads$ ./SigDigger-0.2.0-x86_64-full.AppImage
linux; GNU C++ version 4.8.5 20150623 (Red Hat 4.8.5-16); Boost_105300; UHD_003.010.001.000-0-unknown
erwin@UBU2-VirtualBox:~/Downloads$ 

It flashes a flash window but does not do anything else

I can provide other information as needed
In the meantime I will do a build from source 1804 and or 2004 Ubuntu

de Erwin
PE3ES / F4VTQ

waterwin, 2021-08-09:
Built from source, suggestions for the instructions to make them more helpful for people like me in the attached file, now works with my Pluto+ (Adalm Pluto clone).
[SigDigger-extended build process on Ubuntu 1804.odt](https://github.com/BatchDrake/SigDigger/files/6954627/SigDigger-extended.build.process.on.Ubuntu.1804.odt)

BatchDrake, 2023-04-12:
Hi,

I am closing this issue because we already have a user manual + things changed a lot since 0.2.0. Feel free to open a new issue if you have trouble running the new AppImages.

Cheers,

closed 2023-04-12 by BatchDrake