BatchDrake/SigDigger #159: Hash collision!!! Fatal error!!

phirippu, 2021-08-24:
Hi!

On Ubuntu Hirsute the application does not start with the "Hash collision!!! Fatal error!!" line. 

Qt 5.15.2
Ubuntu 21.04 [linux version 5.11.0-33-lowlatency]
Architecture: x86_64; features: SSE2 SSE3 SSSE3 SSE4.1 SSE4.2 AVX AVX2 RTM HLE

phirippu, 2021-08-24:
Using suscan version 0.2.0 (custom build on Aug 24 2021 at 16:05:24 (10.3.0))
Using sigutils version 0.2.0 (custom build on Aug 24 2021 at 16:04:19 (10.3.0))

BatchDrake, 2023-04-12:
Hi @phirippu 

Sorry for the late answer, I am still going through all open issues today. Things changed a lot since 0.2.0, is SigDigger still failing?

Cheers,

phirippu, 2023-06-01:
No, not anymore =)

BatchDrake, 2023-06-01:
Cool! I am therefore closing the issue, thanks for reporting!

closed 2023-06-01 by BatchDrake