BatchDrake/SigDigger #164: macOS release does not render properly on new MacBook Pros

mehdideveloper, 2021-11-29:
I just ran the latest release (0.3.0 built today) on a clean 2021 MacBook Pro 14" (which uses the Apple silicon, not Intel)
The first issue I notice is that the right pane does not render properly, even in full screen mode, while it works without an issue on the Intel MacBooks.
Here's an screenshot attached. As you see, it's as if the right side needs to be resized, while this is already at max size and covers the whole screen.
I'll let you know if it works properly on the M1 MacBooks (regardless of the UI issue)
[image]

BatchDrake, 2021-11-29:
Hi Mehdi,

Is the side panel width saved after restarting SigDigger?

Cheers,

mehdideveloper, 2021-11-29:
Hi, yes. My changes to the size are saved and properly shown after restarts.

BatchDrake, 2021-11-29:
I believe this one is a minor. I give a 16% default side panel width with
the possibility of adjusting it by hand. Of course, depending on your
window width, this can be more or less.

The problem is that making a guess based on the panel contents somehow does
not work well in all systems (or at least I don't know exactly how). AFAIK
many people faced this issue in the past but nobody provided a useful
solution so far.

mehdideveloper, 2021-12-01:
I can confirm the new build release fixes this issue on M1 MacBook .

BatchDrake, 2021-12-02:
I did not fix anything here. SigDigger is just using the cached side panel size.

mehdideveloper, 2021-12-02:
Aha ok. I had fixed the size on the previous runs.