BatchDrake/SigDigger #165: macOS release does not detect SDRs on M1 MacBooks

mehdideveloper, 2021-11-29:
I tried the latest built on M1 MacBook.
Although SoapySDR detects my SDRs, SigDigger does not. Even clicking on Refresh in the Devices dialog does nothing.
SoapySDR and the modules are compiled from source.

BatchDrake, 2021-11-29:
The .DMG is intended for x86_64 targets. I don't have any M1 machines at
hand to build SigDigger, and I'm surprised you could even execute it. Does
CubicSDR for x86_64 work in your machine?

Cheers,

mehdideveloper, 2021-11-29:
Yes. The new MacBooks (running M1 which itself is ARM-based) run x86 binaries using dynamic binary translation.
I have had the laptop for 4 days now and haven't had time to test SDR apps. I will continue to do my tests and will let you know if I have success in getting it to detect and run SDRs (I may compile it)

mehdideveloper, 2021-11-29:
And regarding your question on CubicSDR: yes I can run CubicSDR without an issue on the M1 mac (I just use their x86 binary)

BatchDrake, 2021-11-29:
Well, I don't think I will be able to debug this issue anytime soon :/

Can you confirm if it works in x86-64 machines? Because if that were the
case, it would look like Apple not being able to emulate their own binaries.

mehdideveloper, 2021-11-29:
No worries. I'll play with the builds and will let you know if I manage to run it.
And no it does not run on my Intel laptop. Actually, when I click on Refresh in the Devices dialog, the app exits (seg fault)
Here's what the debugger shows me:

 thread #13, name = 'QThread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x0000000100b1bed5 libsuscan.0.3.0.dylib`suscan_source_detect_devices + 901
libsuscan.0.3.0.dylib`suscan_source_detect_devices:
->  0x100b1bed5 <+901>: movq   (%r12), %rdi
    0x100b1bed9 <+905>: testq  %rdi, %rdi
    0x100b1bedc <+908>: je     0x100b1bec0               ; <+880>
    0x100b1bede <+910>: leaq   0x2985e(%rip), %rsi       ; "remote"

BatchDrake, 2021-11-29:
Damn, so it is not an M1 problem. I'll look into this tomorrow.

BatchDrake, 2021-11-30:
Just opened this in a Macbook, but I just can't reproduce it. It detects RTL-SDR just fine.

mehdideveloper, 2021-11-30:
You tried on an x86 MacBook, right?
I tried on another x86 MacBook and I don't get the segfault there, but clicking on Refresh does not show the SDR (although SoapySDRUtil detects it)

BatchDrake, 2021-11-30:
I've just tested it back in a macbook, it turns out it had the old libraries installed and I was able to reproduce the crash. I'll look into it these days.

BatchDrake, 2021-12-01:
I may have fixed it. I have already scheduled a build release and you should have a new .dmg within the hour.

Cheers,

mehdideveloper, 2021-12-01:
It actually fixed the other issue (right panel size) but it still does not detect SDRs.

BatchDrake, 2021-12-02:
FYI https://github.com/BatchDrake/SigDigger/issues/99#issuecomment-984535948

mehdideveloper, 2021-12-02:
Great job!
Now it works on my MacBook Pro 14" M1 and detects RTL-SDR.
I will continue to test it with other SDRs.

hpux735, 2021-12-13:
I can verify this works for me on my M1, also.  The rtl-sdr is detected, and works great!

juanibuqt, 2022-01-05:
I tested too in an M1 MacBook Pro and I confirm this new release work great. Thanks!

BatchDrake, 2022-01-05:
I am marking this as fixed, thanks everyone!

closed 2022-01-05 by BatchDrake