BatchDrake/SigDigger #167: BladeRF only runs for a few seconds

hpux735, 2021-12-13:
First of all, thank you so much for making this.  I've made SDR apps in the past, but never one close to as nice as this.  Excellent work.

I'm using a MacBook Pro (with M1), and version 0.3.0.  I've tried with an rtl-sdr, and it seems like I can run it indefinitely.  But, when I use my BladeRF, the processing seems to get stuck or stops after a given quantity of data (some multiple of a buffer?).  If I run it at an 80khz sample rate, I can run it for maybe 30 seconds (a full screen of waterfall on auto).  If I set it to 2.58mhz then it'll run for a few seconds (maybe 2-3), and if I set it to 10mhz it'll run for just a blip.  It seems like it scales exactly inverse-linearly with the sample rate.  This makes me feel like it's processing a single buffer then stopping.

Any ideas?  Thanks!

BatchDrake, 2021-12-13:
Hi William, and thanks for your kind feedback.

Unfortunately, I cannot reproduce any of this. I used to have a BladeRF but
it is now dead. If this is a driver problem, I'm afraid there's little I
can do.

Do you happen have any GNU/Linux machine around? Maybe we can reproduce
under Linux too.

Cheers,

hpux735, 2021-12-13:
I do, yah.  I can try that out.  The best way would be via SigDigger on linux, rather than trying something with SoapySDR?

BatchDrake, 2021-12-13:
Yep, exactly. Try the latest AppImage. If it does not hang, we can try
building SigDigger locally with the latest BladeRF drivers (which are the
ones included in the .dmg) and cross our fingers.

hpux735, 2021-12-13:
Ok, awesome.  I've never tried an AppImage before.  That's incredibly awesome.

Ok, it does still hang in linux, but it helpfully provides some logging...

```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-wdillon'
Qt: XKEYBOARD extension not present on the X server.
[INFO] bladerf_open_with_devinfo()
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:82] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:106] FPGA version (v0.14.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO] bladerf_get_serial() = c52117e2ff2f78588aec8dd89349e432
[INFO] setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
[INFO] setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
[INFO] setSampleRate(Rx, 0, 2.580000 MHz), actual = 2.580000 MHz
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1090] Transfer timed out for buffer 0x2174a20
```

hpux735, 2021-12-13:
I'm just now seeing the version messages.  I have a second BladeRF with older firmware, so I'm willing to load the exact version that libbladeRF wants, if we know what that is.

hpux735, 2021-12-13:
Trying my older BladeRF on linux works!  It also seems to work on my mac now (It hung before, but not now 🤷).  I guess I'll downgrade my firmware for now.  Is there any way to know what commit of libbladerf you used to build so I can see the latest supported versions?

Thanks again!

BatchDrake, 2021-12-13:
Good question, I use whatever is in the latest Ubuntu 18.04 repos. For the
macOS releases I rely on brew, and that supposedly builds everything from
source. I have to look into it.

hpux735, 2021-12-13:
Hm.. Plot thickens.  It seems like Brew is very up-to-date, and is using the 2021.10 release, and is supposed to work with those firmware versions.  Meanwhile ubuntu's is older (2019.07), and is expecting older versions.  I'll load these versions on the "updated" blade, and see what happens...

```
* FPGA bitstream                v0.11.0
* FX3 firmware                  v2.3.2
* libbladeRF                    v2.2.1
* bladeRF-cli                   v1.8.0
* MATLAB & Simulink bindings    v1.0.3
* Python bindings               v1.2.0
```

hpux735, 2021-12-14:
Yep!  I've confirmed that there's a bug in the 2.4.x firmware for the BladeRF.  I even got buffer errors sometimes in bladeRF-cli when I was trying to calibrate it.  I think this can be closed.

Thanks again!

hpux735, 2021-12-14:
And I found this: https://github.com/Nuand/bladeRF/issues/813

closed 2023-04-12 by BatchDrake

hpux735, 2023-04-12:
Fantastic! Can’t wait to try it! ❤️