BatchDrake/SigDigger #168: libsuscan.so.0.2.0: cannot open shared object file: No such file or directory

g4ixt, 2022-01-04:
I followed the build / installation instructions for SigDigger.  Everything appears to have worked except that I get this:

suscan.status
libsuscan.so.0.2.0: cannot open shared object file: No such file or directory

ls -al /usr/local/lib

total 1468
drwxr-xr-x  4 root root    4096 Jan  4 21:00 .
drwxr-xr-x 11 root root    4096 Dec 29 12:10 ..
lrwxrwxrwx  1 root root      16 Jan  4 20:38 libsigutils.so -> libsigutils.so.1
-rw-r--r--  1 root root  429648 Jan  4 20:34 libsigutils.so.0.2.0
lrwxrwxrwx  1 root root      20 Jan  4 20:38 libsigutils.so.1 -> libsigutils.so.0.2.0
lrwxrwxrwx  1 root root      18 Jan  4 21:00 libsuscan.so -> libsuscan.so.0.2.0
-rw-r--r--  1 root root 1055768 Jan  4 21:00 libsuscan.so.0.2.0
drwxr-xr-x  2 root root    4096 Jan  4 21:00 pkgconfig
drwxr-xr-x  3 root root    4096 Oct 12 21:37 python3.9

I thought it might be a $PATH problem so I added /usr/local/lib to the path environment variable but it makes no difference

BatchDrake, 2022-01-04:
You should add /usr/local/lib to LD_LIBRARY_PATH, and not PATH

PS: It seems that you are building from the master branch. You may want to
build everything (sigutils, suscan, SuWidgets and SigDigger) from develop
instead, as it is the branch with the latest changes (including the OpenGL
waterfall).

g4ixt, 2022-01-05:
Thank you.  I already had a .conf file with /usr/local/lib in my /etc/ld.so.conf directory, as per my understanding of Ubuntu set-up, but "echo $LD_LIBRARY_PATH" always gives blank output and I have not yet discovered why.

However, entering in terminal:
export LD_LIBRARY_PATH="usr/local/lib"

and then running suscan.status, fixes the problem and SigDigger now runs (although I guess I will have to do this again after reboot until I figure out how to make it permanent in Ubuntu 20.10)

I will try building the development branch.  Do I just repeat the actions or do i have to uninstall the master branch files somehow first?

BatchDrake, 2022-01-05:
Nah, it is okay if you just pull everything from develop and build from
scratch. The build system is prepared to prevent conflicts with older files.

g4ixt, 2022-01-06:
thanks for the help

closed 2022-01-06 by g4ixt

hoarylea, 2023-04-03:
why wsl cannot run SigDigger?
i do export LD_LIBRARY_PATH="usr/local/lib"

hoarylea, 2023-04-03:
[image]

hoarylea, 2023-04-03:
great you are

hoarylea, 2023-04-03:
> You are probably missing the first `/` in `/usr/local/lib`. `usr/local/lib` is a relative search path, from the current directory. `/usr/local/lib` is the absolute path, referring to where sigutils, suscan and everything you build locally gets installed system-wide.

thank you very much

hoarylea, 2023-04-03:
MR. BatchDrake
i have another question. 
i am a hardware engineer, how could i interface my own fpga hardware to your software.

BatchDrake, 2023-04-03:
I believe it is better off if you open a new issue on that subkect. This one was opened for something else, and I'd prefer to keep threads on-topic.