BatchDrake/SigDigger #169: Building on Linux box

atomicatheist, 2022-01-07:
Here is the output from make:

g++ -s -Wl,-O1 -Wl,-rpath,/home/atomicatheist/anaconda3/lib -o SigDigger AppConfig.o Application.o AppUI.o ColorConfig.o GuiConfig.o Loader.o AudioFileSaver.o AudioPlayback.o GenericAudioPlayer.o AboutDialog.o AudioPanel.o ConfigDialog.o DataSaverUI.o DeviceGain.o DopplerDialog.o FftPanel.o GainSlider.o GenericDataSaverUI.o HistogramDialog.o InspectorPanel.o MainSpectrum.o MainWindow.o PersistentWidget.o SamplerDialog.o SaveProfileDialog.o SourcePanel.o TimeWindow.o Inspector.o InspectorUI.o TVProcessorWorker.o AfcControl.o AskControl.o ClockRecovery.o EqualizerControl.o GainControl.o InspectorCtl.o MfControl.o ToneControl.o AutoGain.o Averager.o Palette.o SNREstimator.o SigDiggerHelpers.o CancellableTask.o ChannelMessage.o GenericMessage.o InspectorMessage.o PSDMessage.o SamplesMessage.o Analyzer.o AnalyzerParams.o Config.o Exception.o Library.o Logger.o Message.o MQ.o SourceInfoMessage.o StatusMessage.o MultitaskController.o Object.o Serializable.o Source.o CarrierDetector.o CarrierXlator.o DopplerCalculator.o HistogramFeeder.o WaveSampler.o AudioMediator.o FftMediator.o InspectorMediator.o PanoramicDialogMediator.o SourceMediator.o SpectrumMediator.o UIMediator.o main.o EstimatorControl.o GenericDataSaver.o FileDataSaver.o SocketForwarder.o NetForwarderUI.o WaitingSpinnerWidget.o DeviceDialog.o DeviceDialogMediator.o PanoramicDialog.o Scanner.o RMSViewer.o RMSViewTab.o RMSViewerSettingsDialog.o LogDialog.o TVProcessorTab.o SymViewTab.o WaveformTab.o MultitaskControllerModel.o BackgroundTasksDialog.o ExportSamplesTask.o AddBookmarkDialog.o BookmarkTableModel.o BookmarkManagerDialog.o TableDelegates.o PortAudioPlayer.o qrc_Icons.o moc_AudioFileSaver.o moc_CarrierDetector.o moc_CarrierXlator.o moc_DopplerCalculator.o moc_DopplerDialog.o moc_GenericDataSaverUI.o moc_HistogramDialog.o moc_HistogramFeeder.o moc_SamplerDialog.o moc_CancellableTask.o moc_Analyzer.o moc_Logger.o moc_MultitaskController.o moc_AboutDialog.o moc_AfcControl.o moc_Application.o moc_AskControl.o moc_AudioPanel.o moc_AudioPlayback.o moc_ClockRecovery.o moc_ConfigDialog.o moc_DataSaverUI.o moc_DeviceGain.o moc_EqualizerControl.o moc_FftPanel.o moc_GainControl.o moc_GainSlider.o moc_Inspector.o moc_InspectorCtl.o moc_InspectorPanel.o moc_InspectorUI.o moc_Loader.o moc_MainSpectrum.o moc_MainWindow.o moc_MfControl.o moc_PersistentWidget.o moc_SaveProfileDialog.o moc_SourcePanel.o moc_TVProcessorWorker.o moc_TimeWindow.o moc_ToneControl.o moc_UIMediator.o moc_EstimatorControl.o moc_GenericDataSaver.o moc_FileDataSaver.o moc_SocketForwarder.o moc_NetForwarderUI.o moc_WaitingSpinnerWidget.o moc_DeviceDialog.o moc_PanoramicDialog.o moc_Scanner.o moc_WaveSampler.o moc_RMSViewer.o moc_RMSViewTab.o moc_RMSViewerSettingsDialog.o moc_LogDialog.o moc_TVProcessorTab.o moc_SymViewTab.o moc_WaveformTab.o moc_MultitaskControllerModel.o moc_BackgroundTasksDialog.o moc_ExportSamplesTask.o moc_AddBookmarkDialog.o moc_BookmarkTableModel.o moc_BookmarkManagerDialog.o moc_TableDelegates.o   -L/home/atomicatheist/anaconda3/lib -lsuwidgets -L/usr/local/lib -lsuscan -lsigutils -lm -lfftw3 -lsndfile -lSoapySDR -lxml2 -lfftw3f -lvolk -lportaudio -lQt5Widgets -lQt5Gui -lQt5Network -lQt5Core -lGL -lpthread
/usr/bin/ld: SourcePanel.o: in function `SigDigger::AutoGain::AutoGain(SigDigger::AutoGain const&)':
SourcePanel.cpp:(.text._ZN9SigDigger8AutoGainC2ERKS0_[_ZN9SigDigger8AutoGainC5ERKS0_]+0x1a6): undefined reference to `std::__throw_bad_array_new_length()'
/usr/bin/ld: SourcePanel.cpp:(.text._ZN9SigDigger8AutoGainC2ERKS0_[_ZN9SigDigger8AutoGainC5ERKS0_]+0x1bb): undefined reference to `std::__throw_bad_array_new_length()'
/usr/bin/ld: Analyzer.o: in function `QtMetaTypePrivate::QMetaTypeFunctionHelper<Suscan::InspectorMessage, true>::Construct(void*, void const*)':
Analyzer.cpp:(.text._ZN17QtMetaTypePrivate23QMetaTypeFunctionHelperIN6Suscan16InspectorMessageELb1EE9ConstructEPvPKv[_ZN17QtMetaTypePrivate23QMetaTypeFunctionHelperIN6Suscan16InspectorMessageELb1EE9ConstructEPvPKv]+0x2df): undefined reference to `std::__throw_bad_array_new_length()'
/usr/bin/ld: Analyzer.cpp:(.text._ZN17QtMetaTypePrivate23QMetaTypeFunctionHelperIN6Suscan16InspectorMessageELb1EE9ConstructEPvPKv[_ZN17QtMetaTypePrivate23QMetaTypeFunctionHelperIN6Suscan16InspectorMessageELb1EE9ConstructEPvPKv]+0x2e9): undefined reference to `std::__throw_bad_array_new_length()'
/usr/bin/ld: Analyzer.cpp:(.text._ZN17QtMetaTypePrivate23QMetaTypeFunctionHelperIN6Suscan16InspectorMessageELb1EE9ConstructEPvPKv[_ZN17QtMetaTypePrivate23QMetaTypeFunctionHelperIN6Suscan16InspectorMessageELb1EE9ConstructEPvPKv]+0x2f3): undefined reference to `std::__throw_bad_array_new_length()'
/usr/bin/ld: /usr/local/lib/libSoapySDR.so: undefined reference to `std::__exception_ptr::exception_ptr::_M_release()@CXXABI_1.3.13'
/usr/bin/ld: /usr/local/lib/libSoapySDR.so: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
/usr/bin/ld: /usr/local/lib/libSoapySDR.so: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'
collect2: error: ld returned 1 exit status
make: *** [Makefile:788: SigDigger] Error 1

Any idea how to resolve these 'undefined reference' errors?

n0p, 2022-01-10:
It would help if you post details about your linux distro and libraries version.

ZadokStrawberry, 2022-01-11:
This must have been an automated email to you? I think this is from my
Ubuntu x64 machine. It’s a wonky install, and I’m using a compiled binary
for sdr on it now.

Thanks,

-//: Larry

atomicatheist, 2022-01-11:
> It would help if you post details about your linux distro and libraries version.

OK, my distro is Pop!_OS 21:10

Here is the output of ldconfig -v:
[libs-list.txt](https://github.com/BatchDrake/SigDigger/files/7850389/libs-list.txt)

ZadokStrawberry, 2022-01-11:
Thanks. I’m running Ubuntu 4.19.0-8-amd64.
I think it was the most recent stable version. when I installed it a few
weeks ago.

Thanks,
-//: Larry Frazier

ZadokStrawberry, 2022-01-11:
I’ll give you the output of ldconfig in a bit. I’m a novice, and tonight
I’ll be listening to the net on this.

atomicatheist, 2022-01-12:
> It looks like it is looking for libstdc++ symbols that just don't exist in your computer. Out of curiosity, do you have multiple C++ library versions installed? Can you provide more details on this "wonky" install?
>
> Alternatively, you can try to download the latest development AppImage (https://github.com/BatchDrake/SigDigger/releases/tag/latest), set execution permissions on (with chmod u+x) and execute it.

Well, the latest AppImage ran OK.
Shame I haven't been able to build from source (even though sigutils, suscan & suwidgets compiled successfully) but this is true for GQRX, CubicSDR & SDR++.  For a novice like me, building from source is hard.
"Half the Captains in Starfleet wouldn't be where they are today if it weren't for Boothby," including Richardson

ZadokStrawberry, 2022-01-13:
I have gcc and gcc arm cross compilers installed, AND Arduino.. I can
compile C and asm native for the machine in question , I didn't check the
md5 on the distro. See the attached. I dont know what "fake root is
supposed to mean. >>>>I may have just "bricked" that laptop with nonsense
distros or something, but I've got another on the way (Old machines have
become a hobby of mine... @#CS , etc,. ) Thanks, -//:Larry Frazier

/usr/lib/x86_64-linux-gnu/libfakeroot:
	libfakeroot-0.so -> libfakeroot-tcp.so
/usr/local/lib:
/lib/x86_64-linux-gnu:
	libnss_systemd.so.2 -> libnss_systemd.so.2
	libnss_myhostname.so.2 -> libnss_myhostname.so.2
	libvo-aacenc.so.0 -> libvo-aacenc.so.0.0.4
	libsystemsettingsview.so.3 -> libsystemsettingsview.so.3
	libXvMCW.so.1 -> libXvMCW.so.1.0.0
	libunwind-coredump.so.0 -> libunwind-coredump.so.0.0.0
	libxerces-c-3.2.so -> libxerces-c-3.2.so
	libnss_mdns_minimal.so.2 -> libnss_mdns_minimal.so.2
	libOkular5Core.so.8 -> libOkular5Core.so.8.0.0
	libpcprofile.so -> libpcprofile.so
	libKF5AkonadiNotes.so.5 -> libKF5AkonadiNotes.so.5.9.3
	libplc4.so -> libplc4.so
	libsmbconf.so.0 -> libsmbconf.so.0
	libtk8.6.so -> libtk8.6.so.0
	libbfd-2.31.1-system.so -> libbfd-2.31.1-system.so
	libksba.so.8 -> libksba.so.8.11.6
	libixml.so.10 -> libixml.so.10.0.1
	libudisks2.so.0 -> libudisks2.so.0.0.0
	libjim.so.0.77 -> libjim.so.0.77
	libshorturlpluginprivate.so.5 -> libshorturlpluginprivate.so.5.9.3
	libiw.so.30 -> libiw.so.30
	libgupnp-igd-1.0.so.4 -> libgupnp-igd-1.0.so.4.2.0
	libortp.so.13 -> libortp.so.13
	libxcb-damage.so.0 -> libxcb-damage.so.0.0.0
	libsoxr.so.0 -> libsoxr.so.0.1.1
	libnss_mdns6.so.2 -> libnss_mdns6.so.2
	libwoff2common.so.1.0.2 -> libwoff2common.so.1.0.2
	libncurses.so.6 -> libncurses.so.6.1
	libkdeinit5_kwin_rules_dialog.so -> libkdeinit5_kwin_rules_dialog.so
	libGeoIP.so.1 -> libGeoIP.so.1.6.12
	libpython3.7m.so.1.0 -> libpython3.7m.so.1.0
	libkdeinit5_ksmserver.so -> libkdeinit5_ksmserver.so
	libCoinMP.so.1 -> libCoinMP.so.1.8.3
	libmariadb.so.3 -> libmariadb.so.3
	libquvi-0.9-0.9.3.so -> libquvi-0.9-0.9.3.so
	liblua5.1-bitop.so.0 -> liblua5.1-bitop.so.0.0.0
	libkdecorations2private.so.6 -> libkdecorations2private.so.5.14.5
	libgps.so.23 -> libgps.so.23.0.0
	libvlc.so.5 -> libvlc.so.5.6.0
	libICE.so.6 -> libICE.so.6.3.0
	libgck-1.so.0 -> libgck-1.so.0.0.0
	libperl.so.5.28 -> libperl.so.5.28.1
	librygel-server-2.6.so.2 -> librygel-server-2.6.so.2.0.4
	libutempter.so.0 -> libutempter.so.1.1.6
	libjansson.so.4 -> libjansson.so.4.11.1
	libdevmapper.so.1.02.1 -> libdevmapper.so.1.02.1
	libargon2.so.1 -> libargon2.so.1
	libuchardet.so.0 -> libuchardet.so.0.0.6
	libhfstospell.so.10 -> libhfstospell.so.10.0.0
	libQt5Xml.so.5 -> libQt5Xml.so.5.11.3
	libiptc.so.0 -> libiptc.so.0.0.0
	libavahi-common.so.3 -> libavahi-common.so.3.5.3
	libnetsnmphelpers.so.30 -> libnetsnmphelpers.so.30.0.3
	libkdeconnectinterfaces.so.1 -> libkdeconnectinterfaces.so.1.3.3
	libgstwebrtc-1.0.so.0 -> libgstwebrtc-1.0.so.0.1404.0
	libKF5GlobalAccelPrivate.so.5 -> libKF5GlobalAccelPrivate.so.5.54.0
	libfluidsynth.so.1 -> libfluidsynth.so.1.7.2
	liblua5.3-socket-unix.so.2 -> liblua5.3-socket-unix.so.2.0.0
	libdv.so.4 -> libdv.so.4.0.3
[... 1295 lines cut ...]
	libntfs-3g.so.883 -> libntfs-3g.so.883.0.0
	libmodplug.so.1 -> libmodplug.so.1.0.0
	libattica.so.0.4 -> libattica.so.0.4.2
	libplasmanm_internal.so -> libplasmanm_internal.so
	libmtp.so.9 -> libmtp.so.9.4.0
	libopus.so.0 -> libopus.so.0.7.0
	libcscreensaver.so.0 -> libcscreensaver.so.0.0.0
	libdcerpc-binding.so.0 -> libdcerpc-binding.so.0.0.1
	libnftnl.so.11 -> libnftnl.so.11.0.0
	libpciaccess.so.0 -> libpciaccess.so.0.11.1
	libtss2-mu.so.0 -> libtss2-mu.so.0.0.0
	libXxf86dga.so.1 -> libXxf86dga.so.1.0.0
	libmplex2-2.1.so.0 -> libmplex2-2.1.so.0.0.0
	libgcrypt.so.20 -> libgcrypt.so.20.2.4
	libWildMidi.so.2 -> libWildMidi.so.2.1.0
/lib:
	libgimpui-2.0.so.0 -> libgimpui-2.0.so.0.1000.8
	libgimpthumb-2.0.so.0 -> libgimpthumb-2.0.so.0.1000.8
	libplasma.so.3 -> libplasma.so.3.0.0
	libkfile.so.4 -> libkfile.so.4.14.38
	libkdeclarative.so.5 -> libkdeclarative.so.5.14.38
	libkwalletbackend.so.4 -> libkwalletbackend.so.4.14.38
	libpurple.so.0 -> libpurple.so.0.13.0
	libbreezecommon4.so.5 -> libbreezecommon4.so.5.14.5
	libknewstuff3.

closed 2023-04-12 by BatchDrake