BatchDrake/SigDigger #177: Connection opened, but host is not a valid Suscan device server

jmangas, 2022-03-17:
Hi
I am having troubles with suscan server (suscli) in raspberrypi and sigdigger client.

[image]

Sigdigger  log:
```
➜  sdr ./SigDigger-0.3.0-x86_64-full.AppImage 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Read type failed!
```
[image]

Suscan log:
`
_pi@raspberrypi:~/sdr $ suscli devserv if=0.0.0.0
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
17 Mar 2022 - 18:03:41 - Suscan device server 0.2.0
17 Mar 2022 - 18:03:41 -   Port 28001: server `Default source'
17 Mar 2022 - 18:03:41 -   Port 28002: server `rtlsdr (Generic RTL2832U OEM :: 00000001)'
17 Mar 2022 - 18:03:41 -   Port 28003: server `rtlsdr (Generic RTL2832U OEM :: 00000001)'
17 Mar 2022 - 18:03:41 -   Port 28004: server `sdrplay (SDRplay Dev0 RSP1A 2103006098)'
17 Mar 2022 - 18:03:41 - Announce server start: 4 profiles
17 Mar 2022 - 18:04:06 - [client 192.168.1.139:41766]: client left
17 Mar 2022 - 18:04:06 - [client 192.168.1.139:41766]: client removed from list (0 outstanding clients)_
`

Please, could you help me?

BatchDrake, 2022-03-17:
Hey, thx for reporting

Could you please attach the output as seen in View->Log messages? That view usually provides more detailed information on what may be going on.

Cheers,

jmangas, 2022-03-17:
Sure. I didn't see that option.
[sigdigger.log](https://github.com/BatchDrake/SigDigger/files/8291653/sigdigger.log)

BatchDrake, 2022-03-17:
Just looked into this. Turns out you are running an incompatible version of
the device server (a rather old version in fact, probably built from
master). Try pulling both sigutils and suscan from the develop branch and
build again.

If you don't feel like compiling again, just make a symlink named suscli to
your SigDigger AppImage, and run this symlink from the command line. It
should behave as a suscli compatible with the AppImage'd SigDigger.

PS: Next week I will be implementing user lists and privileges, which will
break the protocol again. I'll update you as soon as it's done.

Cheers,

jmangas, 2022-03-18:
Hi
Nice trick the suscli symlink. Works perfect in Ubuntu but I can't use that in raspberry. 
Anyway, I compiled again suscan and sigutils from develop branches and it can connect now. 
Although I have another problem. It connects, using sdrplay rsp1a but I don't see nothing en the spectrum. I have to test this again with more love...

Thanks and great work!

jmangas, 2022-03-18:
I set TTL to 100ms and works well now.
Thanks

closed 2022-03-18 by jmangas