BatchDrake/SigDigger #180: Incomplete type 'struct timeval' in QTimeSlider on FreeBSD

jeffpc, 2022-04-11:
I get a build failure on FreeBSD 13 about incomplete type 'struct timeval'.  Adding an include for sys/time.h fixes it.  I don't have time right now to fix it and send in a proper patch that's tested, etc.  I can try to remember to do it tonight, but I figured I'd file a bug since I already mentioned this issue.

```
clang++ -c -pipe -D__FILENAME__=\"SigDigger\" -DHAVE_CURL -O2 -std=gnu++1y -pthread -Wall -Wextra -pthread -fPIC -DQT_DEPRECATED_WARNINGS -DSIGDIGGER_HAVE_PORTAUDIO -D_SU_SINGLE_PRECISION -DHAVE_VOLK=1 -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -Iinclude -I/home/jeffpc/bin/dsp/include/SuWidgets -I/home/jeffpc/bin/dsp/include/suscan -I/home/jeffpc/bin/dsp/include/suscan/util -I/home/jeffpc/bin/dsp/include/sigutils -I/home/jeffpc/bin/dsp/include/sigutils/util -I/usr/local/include -I/usr/local/include/opus -I/usr/local/include/libxml2 -I/usr/local/include/qt5 -I/usr/local/include/qt5/QtOpenGL -I/usr/local/include/qt5/QtWidgets -I/usr/local/include/qt5/QtGui -I/usr/local/include/qt5/QtNetwork -I/usr/local/include/qt5/QtCore -I. -I/usr/local/include -I. -I/usr/local/include -I/usr/local/lib/qt5/mkspecs/freebsd-clang -o MainWindow.o Components/MainWindow.cpp
In file included from Components/MainWindow.cpp:19:
In file included from include/MainWindow.h:24:
include/QTimeSlider.h:32:20: error: field has incomplete type 'struct timeval'
    struct timeval startTime;
                   ^
include/QTimeSlider.h:32:12: note: forward declaration of 'SigDigger::timeval'
    struct timeval startTime;
           ^
include/QTimeSlider.h:33:20: error: field has incomplete type 'struct timeval'
    struct timeval endTime;
                   ^
include/QTimeSlider.h:32:12: note: forward declaration of 'SigDigger::timeval'
    struct timeval startTime;
           ^
2 errors generated.
```

BatchDrake, 2022-04-11:
Don't worry, I'll try to fix this myself. I'll update you as soon as this is fixed.

jeffpc, 2022-04-11:
I see you pushed a fix.  I can confirm that SigDigger now builds without any patches.

closed 2022-04-12 by BatchDrake