BatchDrake/SigDigger #181: RPath Vulnerability in SigDigger suscan v0.2.0 on MacOS (DyLib Hijacker Vulnerability) -

harishperfect, 2022-04-25:
Was noticing something suspicious. And turns out the SigDigger application is vulnerable and an exploitable target by means of the RPath Vulnerability. Detected using DHS (Dylib Hijack Scanner from objective-see_DOT_com)

RFC :) 

Possibly related to [CVE-2008-5516](https://www.opencve.io/cve/CVE-2008-5516)

[image]

BatchDrake, 2022-04-25:
Hey, thanks for reporting!

So, let me get it straight, you managed to hijack a desktop application locally with a regular user, right? Do you have a PoC? Could you describe an exploitation scenario that affects the user, so I can come up with a solution?

On the other hand, how is this related to CVE-2008-5516?

BatchDrake, 2022-10-14:
I am tentatively closing this issue as I don't yet how this can affect the user. Feel free to open it if you find an exploitation scenario.

closed 2022-10-14 by BatchDrake