BatchDrake/SigDigger #182: Crash when opening a RAW IQ-file captured with GQRX

andreascla, 2022-05-18:
When a RAW file with IQ samples captured with GQRX is opened, the following message appears after pressing the "Run" button:

```
Failed to start capture due to errors:
info: Data format detected: 32 bit float
error: Failed to open /Users/andreasc/Downloads/gqrx_20220321_213549_434000000_1200000_fc.raw as raw file: Internal error : SF_INFO struct incomplete.
error: exception in "suscan_source_open_file(new)" (build/CMakeFiles/suscan.dir/compiler_depend.ts:2492)
error: exception in "self->source = suscan_source_new(config)" (build/CMakeFiles/suscan.dir/compiler_depend.ts:508)
error: Failed to initialize source
error: exception in "new->impl = (iface->ctor) (new, ap)" (build/CMakeFiles/suscan.dir/compiler_depend.ts:606)
```

BatchDrake, 2022-05-19:
I have to admit, the more I look into this, the more confused I am. The exerted path explicitly initializes the `SF_INFO` struct as described in `libsndfile`'s documentation, and somehow this never failed until now.

What libsndfile version  are you using? We confirm that this is the latest build from `develop`, right?

Could it be a problem with the file itself? How big is `gqrx_20220321_213549_434000000_1200000_fc.raw`? What is the output of the following command?

```
% ls -l /Users/andreasc/Downloads/gqrx_20220321_213549_434000000_1200000_fc.raw
```

BatchDrake, 2022-05-19:
I've just pushed some changes to suscan that are slightly related to this issue. Try to rebuild suscan again and tell me whether it keeps happening.

andreascla, 2022-05-20:
`libsndfile` is stable 1.1.0 (brew but build from source)
Yes, latest `develop` build.

File is 81 MB (I have tried several files and they all have the same issue):

```
% ls -l /Users/andreasc/Downloads/gqrx_20220321_213549_434000000_1200000_fc.raw
-rw-r--r--@ 1 andreasc  staff    81M Mar 22 13:08 /Users/andreasc/Downloads/gqrx_20220321_213549_434000000_1200000_fc.raw
```

And finally, I rebuilt suscan but still have the same issue. Just to confirm, I don't have to rebuild anything with SigDigger itself, right?

I will try to open them in an x86 Windows machine and see if that makes any difference.

BatchDrake, 2022-05-20:
Unless your SigDigger is particularly old, you wouldn't need to. If you are
not sure, just rebuild to rule that out.

PS I see your file has extended attributes (see the @). What's the output
of ls -l@ gqrx_...?

Cheers,

andreascla, 2022-05-20:
```
% ls -l@ /Users/andreasc/Downloads/gqrx_20220321_213549_434000000_1200000_fc.raw
-rw-r--r--@ 1 andreasc  staff    81M Mar 22 13:08 /Users/andreasc/Downloads/gqrx_20220321_213549_434000000_1200000_fc.raw
	com.apple.macl	  72B
	com.apple.metadata:kMDItemWhereFroms	 147B
	com.apple.quarantine	  57B
```

Good catch, I removed the extended attributes just in case, but same error still...

andreascla, 2022-05-23:
FWIW, the file opens fine on a Windows machine with the latest pre-release (March 28). Is there anything I can do to help debug this issue on Mac?

Flamewires, 2022-08-14:
This is probably related to a recent-ish change to libsndfile to hard cap the max sample rate to 655350

https://github.com/libsndfile/libsndfile/commit/b1d642078f1278f8a1be20ff1f9206415fd34100

BatchDrake, 2022-08-16:
Thanks for the info @Flamewires. I was kind of suspecting something like that. I am triggering a new development build now.

BatchDrake, 2022-10-14:
Quick update: Since many people are still building against the faulty `libsndfile`, I just added a workaround in `develop` to handle raw files with arbitrary sample rates (nothing fancy, I basically trick the library into thinking that the sample rate of the file is 1 ksps). 

This issue has already been addressed in Windows builds thanks to the MSYS2 people, but macOS builds keep failing due to a problem in SoapySDR's PothosCore (I already filed an issue for that, but I have not received any answer yet).

BatchDrake, 2023-04-12:
Hi,

libsndfile seems to have been patched since my last message. Feel free to reopen it if raw files keep crashing after the update.

Cheers,