BatchDrake/SigDigger #183: Allow custom sample rates for Soapy devices

catkira, 2022-07-10:
The Soapy driver returns a list of sample rate, ie SoapyPlutoSDR has this:
```c
std::vector<double> SoapyPlutoSDR::listSampleRates( const int direction, const size_t channel ) const
{
	std::vector<double> options;
	options.push_back(65105);//25M/48/8+1
	options.push_back(1e6);
	options.push_back(2e6);
	options.push_back(3e6);
	options.push_back(4e6);
	options.push_back(5e6);
	options.push_back(6e6);
	options.push_back(7e6);
	options.push_back(8e6);
	options.push_back(9e6);
	options.push_back(10e6);
	return(options);
}
```
However most SDRs can do any sample rate in a given range, also Pluto. Therefore it would be nice, if SigDigger would allow custom sample rates and not only the ones listed by the driver.

catkira, 2022-07-10:
https://github.com/pothosware/SoapyPlutoSDR/issues/53

Some Soapy dev said here, that there is a new API for this: [getSampleRateRange()](https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Device.hpp#L894-L909)

Does SigDigger already support this?

BatchDrake, 2022-07-10:
It does not (yet), but I have already been thinking on how to offer support for that, and the right UI for this. 

I believe I have a solution: How about some checkbox next to the sample rate list (like "manual"), switch back to a line edit box, let the user write it down, and then perform a check to see whether the device supports that range?

Cheers,

catkira, 2022-07-10:
yes, that sounds good!

catkira, 2022-07-10:
the check should only be performed, if the range specification is implemented in the driver.

BatchDrake, 2022-07-31:
I've just added a temptative checkbox to override the sample rate. Build SigDigger from the `develop` branch and tell me whether it works (and, of course, if it saves the setting right after restarting SigDigger, etc)

catkira, 2022-08-01:
There is a weird effect. It seems that the entered custom sample rate gets rounded to the next available sample rate in the list. 

When I enter 12 MSPS and start the stream I get 10 MSPS. When I enter 13 MSPS I get 15 MSPS. When I enter 20 MSPS I get 20 MSPS. When I enter 30 MSPS I get 20 MSPS.

I have manually added 15 MSPS and 20 MSPS to the driver of SoapyPlutoSDR.

BatchDrake, 2022-08-01:
It is weird indeed. Does the GUI revert back to a drop-down list? Can you confirm that you can specify an arbitrary sample rate to the device with other applications?

I'd also like to rule out any UI effect. Put an arbitrary sample rate (one that is clearly not in the list, like, dunno, 12.345678 Mps), click okay in the settings dialog, close SigDigger and attach to this issue the file `~/.suscan/uiconfig.yaml`, so I can see what is actually being saved in the device profile.

Additionally, in your current source directory, locate this file: https://github.com/BatchDrake/SigDigger/blob/develop/Suscan/Analyzer.cpp and right before line 542 (the one with `suscan_analyzer_new`) write this:

```
printf("Analyzer being created with sample rate: %u\n", config.getSampleRate());
```
Compile again, launch SigDigger with a custom rate from the terminal, and tell me what do you see from there once you hit start. If you see your custom sample rate in terminal with no roundups, I am afraid it is actually  the device falling back to a predefined rate. Let's cross our fingers it is not the case 🤞

catkira, 2022-08-01:
The GUI does not revert back.

I guess You mean `~/.suscan/config/uiconfig.yaml`. It seems to be correct, there is `samp_rate: 12345670`

The terminal output is
```
[INFO] Opening PlutoSDR #0 192.168.137.2...
Analyzer being created with sample rate: 12345670
[INFO] Opening PlutoSDR #0 192.168.137.2...
[INFO] Using format CF32.
[INFO] Auto setting Buffer Size: 262144
[INFO] Set MTU Size: 262144
[INFO] Has direct RX copy: 1
```

BatchDrake, 2022-08-01:
I believe I've found the offending lines, but the issue is in Suscan. Locate these highlighted lines in your local Suscan source tree:

https://github.com/BatchDrake/suscan/blob/master/analyzer/source.c#L1757-L1768

And replace them by:

```c
closest_rate = source->config->samp_rate;
```

Then recompile Suscan and install Suscan again. You shouldn't need to recompile SigDigger, just restarting it should be enough.

catkira, 2022-08-01:
Its working now 💯

BatchDrake, 2022-08-01:
Somehow GitHub failed to deliver my last message. Anyways, I am keeping this issue open until I find a clean solution to this issue (most likely testing that the requested sample rate falls in any of the acceptable device ranges). Thanks a lot for your help!

catkira, 2022-08-01:
Yes, if the Soapy driver supports a range, suscan should just test if it falls inside that range.

catkira, 2022-08-30:
@BatchDrake any news about this?

BatchDrake, 2022-08-30:
I am afraid I am blocked these days,  I am still catching up with other
projects and I don't expect to have any spare time for this until the
second half of September (where I will be fixing this and other issues
too).

Sorry for the inconvenience,

BatchDrake, 2023-04-12:
Hi,

I am closing this issue because we already have a way to specifying custom sample rates outside the ranges reported by the devices. Feel free to reopen it if you experience any issues with this feature.

Cheers,

closed 2023-04-12 by BatchDrake