BatchDrake/SigDigger #184: Install SigDigger locally

catkira, 2022-07-10:
Building SigDigger with make and make install as described in README.md does not work for me, because many dependencies are missing (also some include files like compat_time.h etc). 
However building with the scripts/build.sh script works. Is there a simple way how I can install the created binaries from deploy_root locally? Something like make install?

BatchDrake, 2022-07-10:
Hi!

I've just tried building again, and I do see compat-time.h being installed in my system. What operating system do you use? Have you installed sigutils, suscan and SuWidgets in your system? Can your copy your build errors here?

I have detailed build instructions in SigDigger's user manual, check https://github.com/BatchDrake/SigDigger/blob/master/doc/SigDigger_User_Manual.pdf

PS: I see you have opened another issue. Just wanted to let you know that I will be away from my computer for at least a week starting today, and I will not be able to address these issues until Monday 18th. Still, don't hesitate to report as many issues as you find convenient, I'm already planning to do some batch-fixing on my return.

Thanks!

catkira, 2022-07-10:
I am using Ubuntu 20. I get these errors:
```
In file included from include/AudioFileSaver.h:22,
                 from Audio/AudioFileSaver.cpp:20:
include/GenericDataSaver.h:29:10: fatal error: util/compat-time.h: No such file or directory
   29 | #include <util/compat-time.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:2804: AudioFileSaver.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from include/AudioPlayback.h:27,
                 from Audio/AudioPlayback.cpp:21:
include/Suscan/Library.h:36:10: fatal error: sgdp4/sgdp4.h: No such file or directory
   36 | #include <sgdp4/sgdp4.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```

BatchDrake, 2022-07-10:
> I built and installed [strf](https://github.com/cbassa/strf), but that does not install the headers like sgdp4.h.

You don't have to. sigutils provides its own (sane) SGDP4 implementation. I believe the problem is that you have not installed sigutils and suscan first.

BatchDrake, 2022-07-10:
I mean, those files do exist, check them out:

- https://github.com/BatchDrake/sigutils/tree/master/util
- https://github.com/BatchDrake/suscan/tree/master/sgdp4

catkira, 2022-07-10:
yes, but they dont get installed when I do make install
```
benja@DESKTOP:/mnt/d/git/sigutils/build$ sudo make install
Consolidate compiler generated dependencies of target sigutils
[ 67%] Built target sigutils
Consolidate compiler generated dependencies of target sutest
[100%] Built target sutest
Install the project...
-- Install configuration: "Debug"
-- Up-to-date: /usr/local/lib/pkgconfig/sigutils.pc
-- Up-to-date: /usr/local/include/sigutils/sigutils/agc.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/block.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/clock.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/codec.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/coef.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/decider.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/detect.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/equalizer.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/iir.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/lfsr.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/log.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/matfile.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/modem.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/ncqo.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/pll.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/property.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/sampling.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/sigutils.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/smoothpsd.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/softtune.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/specttuner.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/taps.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/tvproc.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/types.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/version.h
-- Up-to-date: /usr/local/include/sigutils/util/util.h
-- Installing: /usr/local/lib/libsigutils.so.0.2.0
-- Up-to-date: /usr/local/lib/libsigutils.so.1
-- Set runtime path of "/usr/local/lib/libsigutils.so.0.2.0" to ""
-- Up-to-date: /usr/local/lib/libsigutils.so
```

BatchDrake, 2022-07-10:
Look:

```
-- Installing: /usr/local/lib/libsigutils.so.0.2.0
-- Up-to-date: /usr/local/lib/libsigutils.so.1
-- Set runtime path of "/usr/local/lib/libsigutils.so.0.2.0" to ""
-- Up-to-date: /usr/local/lib/libsigutils.so
```

It says 0.2.0. This sigutils version is out of date (it should be 0.3.0). Make sure you have the latest sigutils version and try again (the same applies to Suscan).

BatchDrake, 2022-07-10:
For instance, mine prints:

```
(xenon) % sudo make install
[sudo] password for waldo: 
Consolidate compiler generated dependencies of target sigutils
[ 60%] Built target sigutils
Consolidate compiler generated dependencies of target sutest
[100%] Built target sutest
Install the project...
-- Install configuration: "Debug"
-- Up-to-date: /usr/local/lib/pkgconfig/sigutils.pc
-- Up-to-date: /usr/local/include/sigutils/sigutils/agc.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/block.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/clock.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/coef.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/decider.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/defs.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/detect.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/equalizer.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/iir.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/lfsr.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/log.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/matfile.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/ncqo.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/pll.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/property.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/sampling.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/sigutils.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/smoothpsd.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/softtune.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/specttuner.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/taps.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/tvproc.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/types.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/version.h
-- Up-to-date: /usr/local/include/sigutils/sigutils/specific/apt.h
-- Up-to-date: /usr/local/include/sigutils/util/util.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-mman.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-time.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-stat.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-fcntl.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-in.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-inet.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-netdb.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-poll.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-pwd.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-select.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-socket.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-statvfs.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-stdlib.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-termios.h
-- Up-to-date: /usr/local/include/sigutils/util/compat-unistd.h
-- Installing: /usr/local/lib/libsigutils.so.0.3.0
-- Up-to-date: /usr/local/lib/libsigutils.so.1
-- Set runtime path of "/usr/local/lib/libsigutils.so.0.3.0" to ""
-- Up-to-date: /usr/local/lib/libsigutils.so
```

catkira, 2022-07-10:
ah ok, updating sigutils helped. Now I only get this error:
```
In file included from App/Application.cpp:21:
include/Suscan/Library.h:36:10: fatal error: sgdp4/sgdp4.h: No such file or directory
   36 | #include <sgdp4/sgdp4.h>
      |          ^~~~~~~~~~~~~~~
```

BatchDrake, 2022-07-10:
Same procedure for Suscan, yours is probably out of date too.

catkira, 2022-07-10:
I am up to date on master branch

catkira, 2022-07-10:
ah wait, I see the file is being installed...

catkira, 2022-07-10:
ok, now it compiled everything. 
Nevermind, I am a lamer :P

closed 2022-07-10 by catkira