BatchDrake/SigDigger #186: OSX v 0.3.0 - unknown error when trying to open file, errortext not wrapping

sm3ulc, 2022-07-27:
Got this error when opening a file. The error-text does not wrap no not possible find if it's something internal or something with the supplied file.

[image]

BatchDrake, 2022-07-28:
Hey, thanks for reporting.

There is a more detailed log in the log window, press command+L to open it, click the save button and attach the resulting log file to this issue.

Cheers,

grhkm21, 2023-01-10:
Hi, it seems that OP didn't reply. I have come across the same issue, the error window shown above pops up when I hit "play" (?) after loading the file in the settings menu. Here is the log:

```
1673310826.606142,Information,SigDigger,run:82,SigDigger 0.3.0 loaded.
1673310826.701849,Error,source,suscan_source_config_sf_open:405,Failed to open /usr/testsample.wav as audio file: Internal error : SF_INFO struct incomplete.
1673310826.704073,Error,source,suscan_source_config_sf_open:405,Failed to open /usr/testsample.wav as audio file: Internal error : SF_INFO struct incomplete.
1673310841.99339,Error,source,suscan_source_config_open_file_raw:348,Failed to open /usr/testsample.raw as raw file: Internal error : SF_INFO struct incomplete.
1673310841.154203,Error,source,suscan_source_config_open_file_raw:348,Failed to open /usr/testsample.raw as raw file: Internal error : SF_INFO struct incomplete.
1673310843.263638,Error,source,suscan_source_config_open_file_raw:348,Failed to open /usr/testsample.raw as raw file: Internal error : SF_INFO struct incomplete.
1673310843.268283,Error,source,suscan_source_new:2505,exception in "suscan_source_open_file(new)" (build/CMakeFiles/suscan.dir/compiler_depend.ts:2505)
1673310843.271412,Error,local-analyzer,suscan_local_analyzer_source_init:508,exception in "self->source = suscan_source_new(config)" (build/CMakeFiles/suscan.dir/compiler_depend.ts:508)
1673310843.274779,Error,local-analyzer,suscan_local_analyzer_ctor:690,Failed to initialize source
1673310843.277667,Error,analyzer,suscan_analyzer_new_from_interface:606,exception in "new->impl = (iface->ctor) (new, ap)" (build/CMakeFiles/suscan.dir/compiler_depend.ts:606)
```

BatchDrake, 2023-01-10:
Hi,

This issue is related to https://github.com/BatchDrake/SigDigger/issues/182, but long story short: this is a libsndfile issue. While it has already been fixed upstream, it seems that some repositories are still out of date and have faulty versions of the library. Updating libsndfile and rebuilding everything from suscan should do the job.

Cheers,

closed 2023-04-12 by BatchDrake