BatchDrake/SigDigger #188: USRP B205 mini-i Compatibility

jasmitch, 2022-08-30:
Have a USRP B205 connected by USB and SigDigger(AppImage) detects the device however reports it as being unavailable?
I get this message from the application log: "Referenced device `uhd (B205mini 313634B)' is not available. device suscan_source_detect_devices"

I get the following output from the command line:

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

-- Loading firmware image: /usr/share/uhd/images/usrp_b200_fw.hex...
[INFO] Detected Device: B205mini
[INFO] Operating over USB 3.
[INFO] Detected Device: B205mini
[INFO] Operating over USB 3.
[INFO] Detected Device: B205mini
[INFO] Operating over USB 3.
[INFO] Detected Device: B205mini
[INFO] Operating over USB 3.

BatchDrake, 2022-09-15:
Okay, this is odd. Does your USRP work with other applications?

jasmitch, 2022-09-16:
Yeah, it is working fine with gqrx, sdrangel & general USRP utilities.

BatchDrake, 2022-09-16:
When you say it detects the device, but it is not available, where do you see this? In the device window? Or in the configuration dialog? (in which you get to choose the appropriate device) What happens when you click Refresh in the device window?

jasmitch, 2022-09-16:
I get that from the message log(cntrl+l):
"Referenced device `uhd (B205mini 313634B)' is not available. device suscan_source_detect_devices"

When I look at the device windows(cntrl+d), I can see the device but it is greyed out. Pressing the refresh button doesn't
change anything.

BatchDrake, 2022-09-17:
Interesting. Could you do the following?

1. Find the files `~/.suscan/config/uiconfig.yaml` and `~/.suscan/config/sources.yaml` and attach them to this issue (so I can rule out any misconfiguration due to wrong IPs or whatever).
2. Open a terminal window, go to wherever your AppImage is, and softlink it to a file named suscli (and make sure that it is executable). If your AppImage file is named `SigDigger-0.3.0-x86_64-full.AppImage`, you just would need to do:

```
$ ln -s SigDigger-0.3.0-x86_64-full.AppImage suscli
$ chmod a+x suscli
```
3. In that same directory, run `./suscli devices`. You should see something like this:
```
$ ./suscli devices
... (lots of debug logs) ...

 ndx Device name                              Driver   Interface Availability 
------------------------------------------------------------------------------
[ 0] Dummy device                             null     local     unavailable
[ 1] ... (whatever devices you have plugged) ...
```

If I don't find the error here, the next step will be to provide you with an engineering build so we can debug this issue (sounds scary but it is not, it would consist on another SigDigger AppImage that prints more logs and lets you do more tests).

jasmitch, 2022-09-18:
cat ~/..suscan/config/uiconfig.yaml
%TAG ! tag:actinid.org,2022:suscan:
---
- !qtui
  version: 1
  width: 1918
  height: 1111
  x: 1
  y: 60
  fullScreen: false
  sidePanelRatio: 1.536050156739811878e-01
  disableHighRateWarning: false
  loFreq: 35695
  bandwidth: 0
  source: !source_config
    type: SDR
    label: Default source
    interface: local
    freq: 433920000.00000000
    lnb_freq: 0.00000000
    bandwidth: 1000000.00000000
    iq_balance: false
    dc_remove: true
    ppm: 0.00000000
    start_time: 1662178595.229696
    loop: true
    samp_rate: 1000000
    average: 1
    channel: 0
    sdr_args:
      driver: null
    gains: {}
  analyzerParams: !AnalyzerParams
    channelUpdateInterval: 0.10000000
    psdUpdateInterval: 0.04000000
    spectrumAvgAlpha: 0.01000000
    sAvgAlpha: 0.00100000
    nAvgAlpha: 0.50000000
    snr: 2.00000000
    windowSize: 8192
    minFreq: 0.000000000000000000e+00
    maxFreq: 0.000000000000000000e+00
    windowFunction: blackmann-harris
    mode: channel
  colors: !colorcfg
    lcdBackground: '#070d0c'
    lcdForeground: '#3ed180'
    spectrumBackground: '#070d0c'
    spectrumForeground: '#3ed180'
    spectrumAxes: '#152d2b'
    spectrumText: '#ffffff'
    constellationForeground: '#3ed180'
    constellationBackground: '#070d0c'
    constellationAxes: '#152d2b'
    transitionForeground: '#3ed180'
    transitionBackground: '#070d0c'
    transitionAxes: '#152d2b'
    histogramForeground: '#3ed180'
    histogramBackground: '#070d0c'
    histogramAxes: '#152d2b'
    histogramModel: '#ffff00'
    symViewLow: '#000000'
    symViewHigh: '#ffffff'
    symViewBackground: '#134529'
    selection: '#152d2b'
    filterBox: '#327d54'
  guiConfig: !guicfg
    useLMBdrag: false
    noLimits: false
    useGLWaterfall: false
    useMaxBlending: false
    useGlInWindows: false
    enableMsgTTL: true
    msgTTL: 15
  tleSourceConfig: !tlesrccfg
    autoDownloadOnStartup: false
  panoramicSpectrum: !PanoramicDialogConfig
    fullRange: true
    rangeMin: 0.000000000000000000e+00
    rangeMax: 0.000000000000000000e+00
    panRangeMin: -90.00000000
    panRangeMax: 0.00000000
    lnbFreq: 0.000000000000000000e+00
    sampRate: 20000000
    device:
    strategy:
    partitioning:
    palette: Turbo (Gqrx)
  bandPlans: []
  Components:
    AudioWidget: !AudioWidgetConfig
      enabled: false
      collapsed: false
      demod: AM
      rate: 44100
      cutOff: 15000.00000000
      volume: -6.00000000
      savePath:
      squelch: false
      amSquelch: 0.10000000
      ssbSquelch: 0.00100000
      tleCorrection: false
      isSatellite: false
      satName: ISS (ZARYA)
      tleData:
    SourceWidget: !SourceWidgetConfig
      collapsed: false
      throttle: false
      throttleRate: 196000
      dcRemove: true
      iqRev: false
      agcEnabled: false
      gainPresetEnabled: false
      dataSaverConfig: !DataSaverConfig
        path:
      savedPresets: []
    InspToolWidget: !InspToolWidgetConfig
      collapsed: true
      inspectorClass: psk
      inspFactory: GenericInspector
      precise: false
      palette: Suscan
      paletteOffset: 0
      paletteContrast: 0
      autoSquelchTriggerSNR: 10.00000000
    FFTWidget: !FFTWidgetConfig
      collapsed: false
      averaging: 1.00000000
      panWfRatio: 0.30000001
      peakDetect: false
      peakHold: false
      filled: false
      panRangeMin: -60.00000000
      panRangeMax: -10.00000000
      wfRangeMin: -60.00000000
      wfRangeMax: -10.00000000
      palette: Magma (Feely)
      zoom: 1
      rangeLock: true
      timeSpan: 0
      timeStamps: false
      bookmarks: true
      unitName:
      zeroPoint: 0.00000000
      gain: 50.00000000

cat ~/.suscan/config/sources.yaml 
%TAG ! tag:actinid.org,2022:suscan:
---
- !source_config
  type: SDR
  label: Default source
  interface: local
  freq: 433920000.00000000
  lnb_freq: 0.00000000
  bandwidth: 1000000.00000000
  iq_balance: false
  dc_remove: true
  ppm: 0.00000000
  start_time: 1662178595.229696
  loop: true
  samp_rate: 1000000
  average: 1
  channel: 0
  sdr_args:
    driver: null
  gains: {}

./suscli devices
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

 ndx Device name                              Driver   Interface Availability 
------------------------------------------------------------------------------
[ 0] Dummy device                             null     local     available
[ 1] Audio input (hw:HDA Intel PCH,0)         audio    local     available
[ 2] Audio input (default)                    audio    local     available
[ 3] uhd (B205mini 313634B)                   uhd      local     unavailable

I'm currently running Ubuntu 20.04.5 LTS, was thinking of performing a fresh install of 22.04.01 LTS and re-installing the USRP mini driver with relevant packages and trying the AppImage again? 

Would it be better to compile SigDigger from source?

jasmitch, 2022-10-04:
Performed a fresh install of 22.04.01 LTS and installed SigDigger via the dev branch build script at https://actinid.org/blsd.
Can now see the device with this new build and looks to be working fine, thanks for your help.

BatchDrake, 2022-10-04:
Thanks to you! I will link this thread to other users that seem to be affected by the same issue.

BatchDrake, 2022-10-04:
I am also closing it, feel free to reopen it if something related comes up.

closed 2022-10-04 by BatchDrake