BatchDrake/SigDigger #189: Question: Remote operation for SoapyDSR / iio backends

hvegh, 2022-09-02:
Dear Gonzalo,

Just a silly question, I have this [Pluto+](https://github.com/plutoplus/plutoplus) board.

If I hookup the device using USB no issue, SigDigger finds the device and performs great!
I am trying to find out if there is a way on hooking up the device remotely over IP, I have not found out yet.

On the Linux laptop I have both IIO and SoapySDR connecting over IP, see outputs below.

Do I need to install SoapyRemote on the Pluto+, or is there a configuration option I missed?

Kind regards,
Henk

```
$ SoapySDRUtil --probe=driver=plutosdr,hostname=192.168.1.81

[INFO] Opening PlutoSDR #0 192.168.1.81...

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9364
  ad9361-phy,xo_correction=40000000
  backend_version=0.24 (git tag: v0.24)
  fw_version=v0.33-3-gd382-dirty
  hw_model=Analog Devices PlutoSDR Rev.C (Z7010-AD9364)
  hw_model_variant=1
...
```

IIO:
```
$ iio_info -u ip:192.168.1.81 | head

Library version: 0.24 (git tag: d23cbaf)
Compiled with backends: local xml ip usb
IIO context created with network backend.
Backend version: 0.24 (git tag: v0.24)
Backend description string: 192.168.1.81 Linux (none) 5.4.0-gb05d16429dac-dirty #1 SMP PREEMPT Thu Sep 1 00:19:39 CEST 2022 armv7l
IIO context has 9 attributes:
        hw_model: Analog Devices PlutoSDR Rev.C (Z7010-AD9364)
        hw_model_variant: 1
        hw_serial: 1040005401000004ffff0400a0fb533b8d
        fw_version: v0.33-3-gd382-dirty
...
```

BatchDrake, 2022-09-15:
There is, indeed. It is a trick that exists for scenarios like these, in which you want to operate one or more devices remotely, without having to deliver samples at full rate over the network. It involves building sigutils + suscan directly for the Pluto, and configuring remote profiles. You would still need to build the SoapySDR module for Pluto, to be run _inside_ the Pluto. 

The idea is that suscan provides a command line tool named `suscli` which, among many other things, allows you to start the so-called "device server" (`suscli devserv`). Then, from other machine, you connect to this server from SigDigger via TCP and operate with it normally. One of the cool things with it is that you can have more than one SigDigger using the same device from the same or different computers at the same time. Maybe is this what you need?

hvegh, 2022-09-28:
Initially it was just to get access to the IQ via NW.

But what you are suggesting is even better, so we can offload some of the signal processing.
Thank you!

BatchDrake, 2022-10-04:
Damn I completely forgot about this issue. Please check this video: https://www.youtube.com/watch?v=2fJXc9m9SGg

It basically explains how to set up the device server in the remote computer, set the access password, et caetera. Don't hesitate to ask me if you have any doubts.

BatchDrake, 2022-10-04:
Have in mind that this trick does not only work for SDR devices, but  for IQ-based file sources as well.

BatchDrake, 2022-10-21:
Hi @hvegh 

Did this solution work for you?

Cheers,

BatchDrake, 2023-04-12:
Hi,

Since there was not much activity here since last year, I am closing the issue. Feel free to reopen it if you have trouble pulling this off.

Cheers,

closed 2023-04-12 by BatchDrake