BatchDrake/SigDigger #190: SoapySDR Segmentation Fault With USRP

rajgark, 2022-09-12:
Hello, 
I'm trying to get SigDigger up and going and have built everything from source on an Ubuntu 20.04 VM. Everything seems to be running well until I start running after loading the UHD profile. I ran the debug steps with `gdb` and here are the results:

:/usr/local/bin$ gdb ./SigDigger
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./SigDigger...
(No debugging symbols found in ./SigDigger)
(gdb) run
Starting program: /usr/local/bin/SigDigger 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1eb9700 (LWP 25645)]
[New Thread 0x7fffebfff700 (LWP 25646)]
[New Thread 0x7fffeb7fe700 (LWP 25647)]
[New Thread 0x7fffeaffd700 (LWP 25648)]
[New Thread 0x7fffea42e700 (LWP 25650)]
[New Thread 0x7fffe9a0a700 (LWP 25651)]
[New Thread 0x7fffe9209700 (LWP 25652)]
[New Thread 0x7fffe8a08700 (LWP 25653)]
[New Thread 0x7fffcbfff700 (LWP 25654)]
[New Thread 0x7fffcb7fe700 (LWP 25655)]
[New Thread 0x7fffc8ffa700 (LWP 25656)]
[New Thread 0x7fffb594d700 (LWP 25657)]
[Thread 0x7fffb594d700 (LWP 25657) exited]
[New Thread 0x7fffb514c700 (LWP 25658)]
[New Thread 0x7fffb594d700 (LWP 25659)]
[New Thread 0x7fffb494b700 (LWP 25660)]
[Thread 0x7fffb494b700 (LWP 25660) exited]
[New Thread 0x7fffa7fff700 (LWP 25661)]
[New Thread 0x7fffa77fe700 (LWP 25662)]
[New Thread 0x7fffa6ffd700 (LWP 25663)]
[New Thread 0x7fffa67fc700 (LWP 25664)]
[New Thread 0x7fff9ffff700 (LWP 25665)]
[New Thread 0x7fffa57fa700 (LWP 25666)]
[New Thread 0x7fffa5ffb700 (LWP 25668)]
[New Thread 0x7fffa4ff9700 (LWP 25667)]
[New Thread 0x7fff9f7fe700 (LWP 25669)]
[New Thread 0x7fff9effd700 (LWP 25670)]
[New Thread 0x7fff9dffb700 (LWP 25671)]
[New Thread 0x7fff9e7fc700 (LWP 25672)]
[New Thread 0x7fff9d7fa700 (LWP 25673)]
[Thread 0x7fffa7fff700 (LWP 25661) exited]
[New Thread 0x7fff9cff9700 (LWP 25674)]
[Thread 0x7fff9cff9700 (LWP 25674) exited]
[New Thread 0x7fff73fff700 (LWP 25675)]
[Thread 0x7fff73fff700 (LWP 25675) exited]
[Thread 0x7fff9d7fa700 (LWP 25673) exited]
[Thread 0x7fff9dffb700 (LWP 25671) exited]
[Thread 0x7fff9f7fe700 (LWP 25669) exited]
[Thread 0x7fffa57fa700 (LWP 25666) exited]
[Thread 0x7fff9e7fc700 (LWP 25672) exited]
[Thread 0x7fff9ffff700 (LWP 25665) exited]
[Thread 0x7fffa77fe700 (LWP 25662) exited]
[Thread 0x7fffa4ff9700 (LWP 25667) exited]
[Thread 0x7fff9effd700 (LWP 25670) exited]
[Thread 0x7fffa6ffd700 (LWP 25663) exited]
[... 408 lines cut ...]
    port: 0
    user: anonymous
    password:
    mc_if:
    label: User-defined remote profile
  gains: {}
- !source_config
  type: SDR
  label: uhd (Unlabeled device)
  interface: local
  freq: 433000000.00000000
  lnb_freq: 0.00000000
  bandwidth: 0.00000000
  iq_balance: false
  dc_remove: true
  ppm: 0.00000000
  start_time: 1663005230.980343
  loop: true
  samp_rate: 0
  average: 1
  channel: 0
  sdr_args:
    driver: uhd
    host: 10.0.10.141
  gains: {}

BatchDrake, 2022-09-15:
This looks like a driver issue to me. Can you reproduce this crash with other applications too?

rajgark, 2022-09-15:
Hey thanks for the reply. I'm unable to produce this crash with other applications. However, I'm only using Gnu Radio and UHD build otherwise to interact with an Ettus x310 USRP. I was attempting to build SigDigger & all of it's dependencies such that it will work as a Gnu Radio alternative for spectrum analysis. 

When you say driver issue you mean SoapyUHD driver, correct?

BatchDrake, 2022-09-16:
Yes, SoapyUHD. I am asking because, if you see here:

```
0x00007ffff7d66e68 in SoapySDRDevice_setupStream () from /usr/local/lib/libSoapySDR.so.0.8-2
```

The crash happens inside SoapySDR, when the device was already initialized, and the sample stream was being configured. 

There is something suspicious here though. I see the -2 in `libSoapySDR.so.0.8-2`. Could it be that you built SoapySDR by hand, and you are loading the corresponding SoapySDR module from a previous build? Since you are un GNU/Linux, post the output of the following command, so we can start to rule things out:

```
$ find /usr 2> /dev/null | grep SoapySDR | grep '0.8'
```

rajgark, 2022-09-16:
thanks for the reply. Ah okay, yes I see... interesting. I did build SoapySDR by hand, but as far as I know there wasn't a previous build. But, could be because one of the debugging steps I saw was just to recompile? 

I ran the command and unfortunately there was no output :(

rajgark, 2022-09-16:
I did run `find /usr | grep SoapySDR` for an overall view, here is the output for that, if it helps at all:

bhml@bhml-sdr:~$ find /usr | grep SoapySDR
/usr/lib/x86_64-linux-gnu/SoapySDR
/usr/lib/x86_64-linux-gnu/SoapySDR/libSoapyOsmoSDR.so.0.7
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librtlsdrSupport.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libuhdSupport.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libHackRFSupport.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libremoteSupport.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libbladeRFSupport.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libosmosdrSupport.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libRedPitaya.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libLMS7Support.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libairspySupport.so
/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libaudioSupport.so
/usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7
/usr/lib/x86_64-linux-gnu/libSoapySDR.so
/usr/lib/x86_64-linux-gnu/pkgconfig/SoapySDR.pc
/usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7.2
/usr/include/SoapySDR
/usr/include/SoapySDR/Constants.h
/usr/include/SoapySDR/Device.h
/usr/include/SoapySDR/Formats.hpp
/usr/include/SoapySDR/ConverterRegistry.hpp
/usr/include/SoapySDR/Time.h
/usr/include/SoapySDR/Modules.h
/usr/include/SoapySDR/Registry.hpp
/usr/include/SoapySDR/Device.hpp
/usr/include/SoapySDR/Logger.hpp
/usr/include/SoapySDR/Errors.hpp
/usr/include/SoapySDR/Version.h
/usr/include/SoapySDR/Errors.h
/usr/include/SoapySDR/Time.hpp
/usr/include/SoapySDR/Modules.hpp
/usr/include/SoapySDR/Config.h
/usr/include/SoapySDR/Logger.h
/usr/include/SoapySDR/ConverterPrimitives.hpp
/usr/include/SoapySDR/Version.hpp
/usr/include/SoapySDR/Formats.h
/usr/include/SoapySDR/Config.hpp
/usr/include/SoapySDR/Types.h
/usr/include/SoapySDR/Types.hpp
/usr/local/lib/libSoapySDR.so.0.8-2
/usr/local/lib/python2.7/dist-packages/_SoapySDR.so
/usr/local/lib/python2.7/dist-packages/SoapySDR.py
/usr/local/lib/SoapySDR
/usr/local/lib/SoapySDR/modules0.8-2
/usr/local/lib/SoapySDR/modules0.8-2/libuhdSupport.so
/usr/local/lib/libSoapySDR.so.0.8.1
/usr/local/lib/libSoapySDR.so
/usr/local/lib/pkgconfig/SoapySDR.pc
/usr/local/lib/python3.8/dist-packages/_SoapySDR.so
/usr/local/lib/python3.8/dist-packages/SoapySDR.py
/usr/local/include/SoapySDR
/usr/local/include/SoapySDR/Converters.h
/usr/local/include/SoapySDR/Constants.h
/usr/local/include/SoapySDR/Device.h
/usr/local/include/SoapySDR/Formats.hpp
/usr/local/include/SoapySDR/ConverterRegistry.hpp
/usr/local/include/SoapySDR/Time.h
/usr/local/include/SoapySDR/Modules.h
/usr/local/include/SoapySDR/Registry.hpp
/usr/local/include/SoapySDR/Device.hpp
/usr/local/include/SoapySDR/Logger.hpp
/usr/local/include/SoapySDR/Errors.hpp
/usr/local/include/SoapySDR/Version.h
/usr/local/include/SoapySDR/Errors.h
/usr/local/include/SoapySDR/Time.hpp
/usr/local/include/SoapySDR/Modules.hpp
/usr/local/include/SoapySDR/Config.h
/usr/local/include/SoapySDR/Logger.h
/usr/local/include/SoapySDR/ConverterPrimitives.hpp
/usr/local/include/SoapySDR/Version.hpp
/usr/local/include/SoapySDR/Formats.h
/usr/local/include/SoapySDR/Config.hpp
/usr/local/include/SoapySDR/Types.h
/usr/local/include/SoapySDR/Types.hpp
/usr/local/bin/SoapySDRUtil
/usr/local/share/lua/5.1/SoapySDR
/usr/local/share/lua/5.1/SoapySDR/Time.lua
/usr/local/share/lua/5.1/SoapySDR/Logger.lua
/usr/local/share/lua/5.1/SoapySDR/Lib.lua
/usr/local/share/lua/5.1/SoapySDR/Device.lua
/usr/local/share/lua/5.1/SoapySDR/Utility.lua
/usr/local/share/lua/5.1/SoapySDR/init.lua
/usr/local/share/doc/gr-osmosdr/xml/namespaceSoapySDR.xml
/usr/local/share/doc/gr-osmosdr/html/namespaceSoapySDR.html
/usr/local/share/cmake/SoapySDR
/usr/local/share/cmake/SoapySDR/SoapySDRExport-release.cmake
/usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake
/usr/local/share/cmake/SoapySDR/SoapySDRExport.cmake
/usr/local/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
/usr/local/share/cmake/SoapySDR/SoapySDRUtil.cmake
/usr/local/share/man/man1/SoapySDRUtil.1
/usr/share/cmake/SoapySDR
/usr/share/cmake/SoapySDR/SoapySDRConfig.cmake
/usr/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
/usr/share/cmake/SoapySDR/SoapySDRUtil.cmake

BatchDrake, 2022-09-17:
Okay, what happens when you run this? (The full path is important)

```
/usr/local/bin/SoapySDRUtil --probe --args 'driver=uhd'
```

rajgark, 2022-09-19:
Hey sorry for the delay. When I run that, this happens: 

bhml@bhml-sdr:~$ /usr/local/bin/SoapySDRUtil --probe --args 'driver=uhd'
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.3.0.0-release
[INFO] [x300_impl.cpp:212] [X300] X300 initialization sequence...
[INFO] [X300] X300 initialization sequence...
[INFO] [x300_eth_mgr.cpp:648] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Maximum frame size: 1472 bytes.

However, it gets stuck here and does not successfully probe the USRP. Been stuck at this for about 3-4 minutes now. Interesting, so this must be the error in initializing the stream that results in the segmentation fault...

rajgark, 2022-09-19:
interestingly enough, when I run `/usr/local/bin/SoapySDRUtil --probe` without driver specified, I get a successful probe (same with `uhd_usrp_probe`):

bhml@bhml-sdr:~$ /usr/local/bin/SoapySDRUtil --probe 
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.3.0.0-release
[INFO] [x300_impl.cpp:212] [X300] X300 initialization sequence...
[INFO] [X300] X300 initialization sequence...
[INFO] [x300_eth_mgr.cpp:648] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [x300_impl.cpp:413] [X300] Radio 1x clock: 200 MHz
[INFO] [X300] Radio 1x clock: 200 MHz

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=x300
  hardware=X310
  fpga_version=39.0
  fw_version=6.0
  mboard_id=X310
  mboard_name=
  mboard_serial=3253FAC
  rx0_antenna=RX1
  rx0_id=TwinRX Rev C (0x0095)
  rx0_ref_power_key=x3xx_pwr_twinrx_rx_rx1
  rx0_ref_power_serial=3247771#0
  rx0_serial=3247771
  rx0_subdev_name=TwinRX RX0
  rx0_subdev_spec=A:0 A:1 B:0 B:1
  rx1_antenna=RX2
  rx1_id=TwinRX Rev C (0x0095)
  rx1_ref_power_key=x3xx_pwr_twinrx_rx_rx2
  rx1_ref_power_serial=3247771#1
  rx1_serial=3247771
  rx1_subdev_name=TwinRX RX1
  rx1_subdev_spec=A:0 A:1 B:0 B:1
  rx2_antenna=RX1
  rx2_id=TwinRX Rev C (0x0095)
  rx2_ref_power_key=x3xx_pwr_twinrx_rx_rx1
  rx2_ref_power_serial=31E878F#0
  rx2_serial=31E878F
  rx2_subdev_name=TwinRX RX0
  rx2_subdev_spec=A:0 A:1 B:0 B:1
  rx3_antenna=RX2
  rx3_id=TwinRX Rev C (0x0095)
  rx3_ref_power_key=x3xx_pwr_twinrx_rx_rx2
  rx3_ref_power_serial=31E878F#1
  rx3_serial=31E878F
  rx3_subdev_name=TwinRX RX1
  rx3_subdev_spec=A:0 A:1 B:0 B:1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 4 Rx, 0 Tx
  Timestamps: YES
  Clock sources: internal, external, gpsdo
  Time sources: internal, external, gpsdo
  Sensors: ref_locked
     * ref_locked (Ref): true locked
  GPIOs: FP0A, RXA, TXA, FP0B, RXB, TXB

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
[... 126 lines cut ...]
       [key=peak, default=1.0, type=float]
     * Receive socket buffer size - The size of the kernel socket buffer in bytes. Use 0 for automatic.
       [key=recv_buff_size, units=bytes, default=0, type=int]
     * Receive frame buffer size - The size an individual datagram or frame in bytes.
       [key=recv_frame_size, units=bytes, type=int]
     * Receive number of buffers - The number of available buffers.
       [key=num_recv_frames, units=buffers, type=int]
     * Full-scale amplitude - Specifies the full-scale amplitude when using floats (not supported for all devices).
       [key=fullscale, default=1.0, type=float]
  Antennas: RX1, RX2
  Full gain range: [0, 99, 0.5] dB
    ADC-digital gain range: [0, 6, 0.5] dB
    all gain range: [0, 93, 1] dB
  Full freq range: [0, 6040] MHz
    RF freq range: [0, 6040] MHz
    BB freq range: [-100, 100] MHz
  Tune args:
     * N divider - Whether the daughterboard tune code should use an integer N divider or fractional N divider (not supported for all devices).
       [key=mode_n, type=string, options=(integer, fractional)]
     * Integer-N tuning step - The step between valid tunable frequencies when using integer-N tuning (not supported for all devices).
       [key=int_n_step, type=float]
  Sample rates: 0.196078, 0.19685, 0.197628, 0.198413, 0.199203, ..., 40, 50, 66.6667, 100, 200 MSps
  Filter bandwidths: 80 MHz
  Sensors: lo_locked
     * lo_locked (LO): true locked

BatchDrake, 2022-09-19:
This is quite interesting. If you see, the driver used SoapySDRUtil is not UHD, but x300 instead.

Let's try this: in the source config dialog, in the device tweaks button, add an entry with key `driver` and value `x300`, accept, save, and start the capture. Let's see whether that crashes.

rajgark, 2022-09-19:
Oh yes interesting... okay sounds good. I'm trying to find the source config file, is it ~/.suscan/config/sources.yaml which I specify this in?

rajgark, 2022-09-19:
Okay so I loaded an existing profile for the USRP and in the SDR config dialog I changed the driver from uhd to x300 and this is the following message after acquisition:

Failed to start capture due to errors:
error: Failed to open SDR device: SoapySDR::Device::make() no match
error: exception in "suscan_source_open_sdr(new)" (analyzer/source.c:2513)
error: exception in "self->source = suscan_source_new(config)" (analyzer/impl/local.c:508)
error: Failed to initialize source
error: exception in "new->impl = (iface->ctor) (new, ap)" (analyzer/analyzer.c:6

BatchDrake, 2022-09-19:
Actually, you can do it from the GUI, there's a button named "device specific tweaks" for this purpose in the config dialog. 

If you prefer to edit config files directly, the one you need is uiconfig.yaml. That's where the default UI profile is stored.

rajgark, 2022-09-19:
I see, that is easier. So the error is this: 

Failed to start capture due to errors:
error: Failed to open SDR device: SoapySDR::Device::make() no match
error: exception in "suscan_source_open_sdr(new)" (analyzer/source.c:2513)
error: exception in "self->source = suscan_source_new(config)" (analyzer/impl/local.c:508)
error: Failed to initialize source
error: exception in "new->impl = (iface->ctor) (new, ap)" (analyzer/analyzer.c:6

And I'll attach the pictures of the config: 
[image]

[image]

[image]

BatchDrake, 2022-09-19:
Try setting these keys too:

```
driver=x300
hardware=X310
fpga_version=39.0
fw_version=6.0
mboard_id=X310
mboard_name=
mboard_serial=3253FAC
```

rajgark, 2022-09-19:
Some progress, after setting the keys I re-tried acquisition and the following happened:

[INFO] [x300_impl.cpp:212] [X300] X300 initialization sequence...
[INFO] [X300] X300 initialization sequence...
[INFO] [x300_eth_mgr.cpp:648] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [x300_impl.cpp:413] [X300] Radio 1x clock: 200 MHz
[INFO] [X300] Radio 1x clock: 200 MHz
[INFO] [x300_impl.cpp:212] [X300] X300 initialization sequence...
[INFO] [X300] X300 initialization sequence...
[INFO] [x300_eth_mgr.cpp:648] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [x300_impl.cpp:413] [X300] Radio 1x clock: 200 MHz
[INFO] [X300] Radio 1x clock: 200 MHz
[WARNING] [multi_usrp_rfnoc.cpp:1245] [MULTI_USRP] Could not set RX rate to 0.196 MHz. Actual rate is 0.196 MHz
[WARNING] [MULTI_USRP] Could not set RX rate to 0.196 MHz. Actual rate is 0.196 MHz
Segmentation fault (core dumped)

BatchDrake, 2022-09-19:
I've just added a few more changes to Suscan, regarding sample rate stuff. Can you rebuild everything from scratch (sigutils, suscan, SuWidgets and SigDigger, in this order) and, in case this keeps happening, post a stack trace?

rajgark, 2022-09-19:
Yes will do. Wow that's great, I seriously greatly appreciate this thank you so much

rajgark, 2022-09-20:
got it, here is the stack trace: 

bhml@bhml-sdr:/usr/local/bin$ gdb ./SigDigger
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./SigDigger...
(No debugging symbols found in ./SigDigger)
(gdb) run
Starting program: /usr/local/bin/SigDigger 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1f1d700 (LWP 40237)]
[New Thread 0x7fffebfff700 (LWP 40238)]
[New Thread 0x7fffeb7fe700 (LWP 40239)]
[New Thread 0x7fffeaffd700 (LWP 40240)]
[New Thread 0x7fffea490700 (LWP 40242)]
[New Thread 0x7fffe9a6c700 (LWP 40243)]
[New Thread 0x7fffe926b700 (LWP 40244)]
[New Thread 0x7fffe8a6a700 (LWP 40245)]
[New Thread 0x7fffcbfff700 (LWP 40246)]
[New Thread 0x7fffcb7fe700 (LWP 40247)]
[New Thread 0x7fffc9059700 (LWP 40248)]
[New Thread 0x7fffb594d700 (LWP 40249)]
[New Thread 0x7fffb514c700 (LWP 40250)]
[Thread 0x7fffb594d700 (LWP 40249) exited]
[New Thread 0x7fffb594d700 (LWP 40251)]
[New Thread 0x7fffb494b700 (LWP 40252)]
[New Thread 0x7fffaffff700 (LWP 40253)]
[Thread 0x7fffb494b700 (LWP 40252) exited]
[New Thread 0x7fffaf7fe700 (LWP 40254)]
[New Thread 0x7fffaeffd700 (LWP 40255)]
[New Thread 0x7fffae7fc700 (LWP 40256)]
[New Thread 0x7fffadffb700 (LWP 40257)]
[New Thread 0x7fffad7fa700 (LWP 40258)]
[New Thread 0x7fffacff9700 (LWP 40259)]
[New Thread 0x7fff9bfff700 (LWP 40260)]
[New Thread 0x7fff9affd700 (LWP 40262)]
[New Thread 0x7fff9b7fe700 (LWP 40261)]
[New Thread 0x7fff9a7fc700 (LWP 40263)]
[Thread 0x7fff9b7fe700 (LWP 40261) exited]
[New Thread 0x7fff99ffb700 (LWP 40264)]
[Thread 0x7fffaffff700 (LWP 40253) exited]
[New Thread 0x7fff997fa700 (LWP 40265)]
[New Thread 0x7fff98ff9700 (LWP 40266)]
[New Thread 0x7fff87fff700 (LWP 40267)]
[Thread 0x7fff9a7fc700 (LWP 40263) exited]
[Thread 0x7fffaeffd700 (LWP 40255) exited]
[Thread 0x7fff87fff700 (LWP 40267) exited]
[Thread 0x7fffacff9700 (LWP 40259) exited]
[Thread 0x7fffad7fa700 (LWP 40258) exited]
[Thread 0x7fffae7fc700 (LWP 40256) exited]
[Thread 0x7fffaf7fe700 (LWP 40254) exited]
[Thread 0x7fff997fa700 (LWP 40265) exited]
[Thread 0x7fff9affd700 (LWP 40262) exited]
[Thread 0x7fffadffb700 (LWP 40257) exited]
[Thread 0x7fff98ff9700 (LWP 40266) exited]
[Thread 0x7fffb594d700 (LWP 40251) exited]
[... 413 lines cut ...]
#1  0x00007ffff4ffd36e in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff4ffd4a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff4ffd4f1 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff5026ad1 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff664e609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#6  0x00007ffff6227133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 2 (Thread 0x7ffff1f1d700 (LWP 40237)):
#0  0x00007ffff621a99f in __GI___poll (fds=0x7ffff1f1c928, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff3116c1a in  () at /lib/x86_64-linux-gnu/libxcb.so.1
#2  0x00007ffff311890a in xcb_wait_for_event () at /lib/x86_64-linux-gnu/libxcb.so.1
#3  0x00007ffff2a30298 in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#4  0x00007ffff67299d2 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#5  0x00007ffff664e609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#6  0x00007ffff6227133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 1 (Thread 0x7ffff30dc000 (LWP 40233)):
#0  0x00007ffff621a99f in __GI___poll (fds=0x5555568baf80, nfds=5, timeout=110) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff4ffd36e in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff4ffd4a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff6949435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007ffff68f03ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff68f8116 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x0000555555620ce6 in main ()

BatchDrake, 2022-09-20:
Look at this:

```
Thread 53 (Thread 0x7fffae7fc700 (LWP 40291)):
#0 0x00007ffff621a99f in __GI___poll (fds=0x7fffae7faca0, nfds=1, timeout=100) at ../sysdeps/unix/sysv/linux/poll.c:29
https://github.com/BatchDrake/SigDigger/issues/1 0x00007fffb64a3b22 in uhd::transport::recv_link_baseuhd::transport::udp_boost_asio_link::get_recv_buff(int) () at /lib/libuhd.so.4.3.0
```

Whatever is happening, it is happening inside `libuhd.so.4.3.0`. It is stuck there. Let's get back to the applications in your system that actually work. For instance, GnuRadio, what is the UHD source you are using? Can you verify that GnuRadio is also using `libuhd.so.4.3.0`? (this is easy, just open GnuRadio companion with gdb, start the UHD source, hit Ctrl+C, `thread apply all bt` and see what libraries are being used).

On the other hand, I am also interested in the `x300` crash. Can you `gdb/thread apply all bt` it?

PD: If nothing of this is working, we can always resort to more aggresive measures. Is there the possibility that you set up a SSH server with an unprivileged user so I can connect to it remotely and attempt to fix it? Preferably through a VM, of course.

Thanks!

rajgark, 2022-09-21:
Oh I see... Got pulled in to some other work, I will try this later today and post an update. Again, thank you so much really

BatchDrake, 2022-10-04:
Hi again,

Other user was also having trouble with SigDigger + USPR. His approach was slightly different, he used [blsd](https://actinid.org/blsd), which is a script I wrote some time ago to automatize the download and build of SigDigger. See https://github.com/BatchDrake/SigDigger/issues/188

Could you download & run blsd and tell me whether it keeps failing? Since it is kind-of a standarized build process, it could provide me with a base line for further debugging steps.

Note that in this case you do not need to install anything system-wide. Everything is deployed in a subdirectory of the current working directory, from which you can run SigDigger directly (using ./SigDigger).

alphafox02, 2022-11-04:
I have an X310 and have used it with SigDigger. I'll check it again tonight. I'm using the UHD 4.x that's provided with Lubuntu 22.04 (DragonOS). If I check and it's working for me, maybe you could boot and run it live and see if it works for you, that might narrow it down to something with your install.