BatchDrake/SigDigger #191: Installing SidDigger on Windows....

Lowlay75, 2022-10-26:
Hi all,
I'm not very computer savvy and would like to use SigDigger on my Windows PC.
However despite all the explanations I do not understand how to install it.
I am also new to SDR.
Thanking you.

BatchDrake, 2022-10-26:
Hi,

Support for Windows is still in experimental state, it only works with
RTL-SDR and it is currently crashing for reasons I'm struggling to debug.
So, for the time being, I do not have a working release for Windows (yet).
I can update you as soon as I have these issues fixed.

On the other hand, if you happen to have a GNU/Linux box, you can try the
AppImage instead. It supports lots of devices and it is generally faster
than the Windows release.

Cheers,

Lowlay75, 2022-10-26:
Thank you for your message ! 
I understand, I would then wait impatiently for its release on Windows

closed 2022-10-26 by Lowlay75