BatchDrake/SigDigger #192: Use a standard binary name in Linux

drws, 2022-10-27:
_SigDigger_ currently builds and installs as a `*/bin/SigDigger`. The problem is in the capitalized binary name, which is not standard Linux practice. Please rename the binary to `sigdigger` at least for Linux systems.

BatchDrake, 2022-10-27:
Hi,

This is actually easy to change. However, I am struggling to find any applicable standard describing this case restriction (which, by the way, would change what people already expect to find in the $PATH in a future release). Could you point me to the right document so I can give it a look?

Thanks,

drws, 2022-10-27:
It's not an absolute restriction and I'm not sure it's written down in any currently widely accepted standard. I checked the [latest FHS](https://wiki.linuxfoundation.org/lsb/fhs-30), but couldn't find it. That's also why I somewhat loosely described it as _a standard practice_.

I'm not sure I get your reference to the $PATH. Even FHS only lists lowercase paths, but doesn't provide a general restriction, so lowercase (sub)paths are only implied. Some opensource tools supported by big companies do use capitalized paths and names, but that's probably not what you had in mind.

BatchDrake, 2022-10-28:
Honestly, the only thing I had in mind was having an executable name that matched that of the software, and was the same across platforms (FreeBSD, macOS, Windows?)

I am marking this as minor, as I am failing to see the urgency in changing the executable name, at least for now. I wouldn't be surprised if this ends up being a requirement in order to be in the official repos of distros like Debian, though!

Cheers,

drws, 2022-10-28:
Since Windows needs the `.exe` it won't let you have matching names in any case, except if all three binaries were called `SigDigger.exe`. So the logic can be rather reversed and binaries could have distincitve names given the platform. It's the pragmatic solution since package and binary names are OS-dependent and (un)fortunately that's just the way it is. I'm afraid that anything else is _aesthetics over functionality_.

jeffpc, 2022-10-28:
There is software that gets packaged with non-"standard" names, so while it may not be common, it is certainly not out of the ordinary.  I found these in my $PATH:

* GraphicsMagick*
* HandBrake
* NanoVNASaver
* R / Rscript
* SoapySDR*
* Xorg / Xvnc

With all that said, I don't really care what the binary is called. :)

closed 2023-04-12 by BatchDrake