BatchDrake/SigDigger #193: Provide a .desktop file for Linux systems

drws, 2022-10-27:
Since SigDigger is a GUI application it could also provide a desktop icon. In Linux systems this can be achieved with a `.desktop` file. My current one for SigDigger contains:

    #!/usr/bin/env xdg-open
    [Desktop Entry]
    Version=1.0
    Type=Application
    Name=SigDigger
    Exec=SigDigger
    Icon=application-executable
    Categories=HamRadio;

BatchDrake, 2022-10-27:
Hi,

This is something already exists. Line 57 in SigDigger's project file (https://github.com/BatchDrake/SigDigger/blob/master/SigDigger.pro#L57) installs it explicitly in `$PREFIX/share/applications`, and it is from this input file: https://github.com/BatchDrake/SigDigger/blob/master/SigDigger.desktop.in

Maybe this is not working in your case?

Cheers,

BatchDrake, 2022-10-29:
Hi,

Have you located the .desktop file?

Cheers,

drws, 2022-11-02:
The issue is in the Arch Linux [package](https://archlinux.org/packages/community/x86_64/sigdigger/). If you check its [file list](https://archlinux.org/packages/community/x86_64/sigdigger/files/) it shows the error is clearly at their side. I've filed a [relevant report](https://bugs.archlinux.org/task/76381). Thank you for your help!

closed 2022-11-02 by drws