BatchDrake/SigDigger #195: configure  succeeds and then the build fails with the missing WFHelpers.h header

yurivict, 2023-01-07:
```
include/Suscan/Library.h:37:10: fatal error: 'WFHelpers.h' file not found
#include <WFHelpers.h>
         ^~~~~~~~~~~~~
```

BatchDrake, 2023-01-08:
Hi Yuri,

Is this error still occurring? How are you compiling SigDigger exactly? Can
you detail the steps you followed?

Cheers,

yurivict, 2023-01-08:
I use qmake.

BatchDrake, 2023-01-08:
Hi,

Still, this should work out of the box. Which version in particular? Is
this Qt5 or Qt6?

You are building this on FreeBSD right? Which version in particular?

Cheers,

yurivict, 2023-01-08:
It's Qt5.

The original error occurred when suwidgets was missing.
Once suwidgets is installed ```-I${LOCALBASE}/include/SuWidgets``` had to be added in order for SigDigger to find the WFHelpers.h header.

yurivict, 2023-01-08:
It's FreeBSD 13.1 STABLE

chron0, 2023-03-13:
I have the same issue with gentoo and 0.3.0:

```
_WARNINGS -DSIGDIGGER_HAVE_ALSA -D_SU_SINGLE_PRECISION -DHAVE_VOLK=1 -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -Iinclude -I/usr/include/qt5/SuWidgets -I/usr/include/suscan -I/usr/include/suscan/util -I/usr/include/sigutils -I/usr/include/sigutils/util -I/usr/include/opus -I/usr/include/libxml2 -I/usr/include/qt5 -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtCore -I. -I/usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o AppUI.o App/AppUI.cpp
In file included from App/Application.cpp:21:
include/Suscan/Library.h:35:10: fatal error: analyzer/discovery.h: No such file or directory
   35 | #include <analyzer/discovery.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:2832: Application.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from include/AppConfig.h:29,
                 from App/AppConfig.cpp:20:
include/Version.h:23:10: fatal error: sigutils/version.h: No such file or directory
   23 | #include <sigutils/version.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:2795: AppConfig.o] Error 1
In file included from App/AppUI.cpp:22:
include/MainSpectrum.h:26:10: fatal error: WFHelpers.h: No such file or directory
   26 | #include <WFHelpers.h>
      |          ^~~~~~~~~~~~~
```

0.1.0 built without a problem - but neither suwidgets nor suscan deployed a WFHelpers.h during install and there is none /usr/*

BatchDrake, 2023-03-13:
Make sure that you are building the latest WFHelpers.h from develop. The
same applies to Sigutils and Suscan.

chron0, 2023-03-13:
yeah, i think thats the issue, i was hunting down the headers and it seems the are only in master but not in 0.1 release on suscan and suwidgets

chron0, 2023-03-13:
this was my conclusion after looking thru master branch and finding a lot of files that simply werent in my system and also in the 0.1.0 release package. I have created local ebuilds in my overlay and bumped sigutils, suscan and suwidgets to -9999 (git master) and they compiled without a problem. consequently my sigdigger-0.3.0 now also compiled perfectly. 

I think it would make sense to bump sigutils, suscan and suwidgets to a 0.3.0 release where they match so that package and build maintainers for the distros can pick it up and avoid this dep confusion.

BatchDrake, 2023-03-13:
Yeah, you are absolutely right. I got into this bad habit of only caring about SigDigger versioning, but this is something I should do in all projects at the same time.

chron0, 2023-03-13:
and ofc, big thank you for putting in all the work and sharing it!

closed 2023-04-12 by BatchDrake