BatchDrake/SigDigger #197: Build fails because both SuWidgets and SigDigger have the file Version.h

yurivict, 2023-01-08:
```
--- SigDiggerHelpers.o ---
Misc/SigDiggerHelpers.cpp:53:18: error: use of undeclared identifier 'SIGDIGGER_VERSION_STRING'
  return QString(SIGDIGGER_VERSION_STRING);
                 ^
1 error generated.
```

Adding ```-I${srcdir}/include``` before all other options helps, but you should rename Version.h files to prevent this conflict.

BatchDrake, 2023-01-08:
Hi Yuri,

Can you detail what your system is, which compiler & version are you using,
and what Qt version are you linking?

Thanks,

BatchDrake, 2023-01-08:
Also, I'd like to know how what steps you are following in order to compile. Version.h are two different files that must exist in SuWidgets AND SigDigger, and the build system should tell them apart.

yurivict, 2023-01-08:
I build with qmake.

First, I get this error:
```
In file included from Components/QTimeSlider.cpp:27:
In file included from include/SigDiggerHelpers.h:24:
include/Suscan/Library.h:37:10: fatal error: 'WFHelpers.h' file not found
#include <WFHelpers.h>
         ^~~~~~~~~~~~~
```
Once the path to WFHelpers.h is added it fails because the same folder where WFHelpers.his also has Version.h

BatchDrake, 2023-01-08:
I must admit, I am a bit confused. SigDigger has been built in SigDigger in
the past, but I believe we never faced this issue earlier. In fact,
SigDigger and SuWidgets' Version.h should never conflict because they are
in different subdirectories.

How did you install SuWidgets? make install after qmake? Can you paste here
the output of the make install?

PS: for quick build and deployment I created a script named blsd (
https://actinid.org/blsd) some time ago, that downloads SigDigger and
builds it inside a user directory, without touching anything system-wide. I
must admit, I never tried it in FreeBSD, can you tell me whether it works
for you?

Cheers,

yurivict, 2023-01-08:
https://people.freebsd.org/~yuri/suwidgets-0.1.0-147.log

I only use FreeBSD ports framework to build software. This leads to reproducible, standardized builds.

BatchDrake, 2023-01-08:
That is excellent news, because I believe I can setup a system identical to
yours and attempt to reproduce your errors step by step. Do you have a link
at hand to a CD / disk I can use to setup a VM?

Cheers,

yurivict, 2023-01-08:
1. Install the [FreeBSD VM image](https://download.freebsd.org/ftp/releases/VM-IMAGES/13.1-RELEASE/amd64/Latest/) into VirtualBox.
2. Boot FreeBSD
3. Install git: pkg install git
4. Check out the ports tree: git clone https://git.FreeBSD.org/ports.git /usr/ports
5. cd /usr/ports/cad/comms/sigdigger
6. Install dependencies: pkg install -A `make missing | grep -v suwidgets | grep -v suscan | grep -v sigutils`
7. Build: make 
8. Run tests: make test

jeffpc, 2023-01-08:
FWIW, I have my own script to build SigDigger locally.  I does the bare minimum without any fancy stuff like @BatchDrake's script.  Just FWIW. 

I just ran it on the latest develop branch, and everything builds ok.

I also tried to build the sigdigger port on my other fbsd 13.1 box just now (cd /usr/ports/comms/sigdigger; make) and all 4 packages successfully built.

```
#!/bin/sh

TARGET_DIR=$HOME/bin/dsp
QMAKE=qmake-qt5
MAKEARGS=-j4

set -e
set -x

cd sigutils
rm -rf build
mkdir build
cd build
PKG_CONFIG_PATH=$TARGET_DIR/lib/pkgconfig/ cmake .. \
	-DCMAKE_INSTALL_PREFIX=$TARGET_DIR \
	-DCMAKE_BUILD_TYPE=Release
make $MAKEARGS
make install
cd ../..

cd suscan
rm -rf build
mkdir build
cd build
PKG_CONFIG_PATH=$TARGET_DIR/lib/pkgconfig/ cmake .. \
	-DCMAKE_INSTALL_PREFIX=$TARGET_DIR \
	-DCMAKE_BUILD_TYPE=Release
make $MAKEARGS
make install
cd ../..

cd SuWidgets
PKG_CONFIG_PATH=$TARGET_DIR/lib/pkgconfig/ \
	$QMAKE SuWidgetsLib.pro \
	"CONFIG += release" \
	PREFIX=$TARGET_DIR
make $MAKEARGS
make install
cd ..

cd SigDigger
PKG_CONFIG_PATH=$TARGET_DIR/lib/pkgconfig/ \
	$QMAKE SigDigger.pro \
	"CONFIG += release" \
	SUWIDGETS_PREFIX=$TARGET_DIR \
	PREFIX=$TARGET_DIR
make $MAKEARGS
make install
cp SigDigger $TARGET_DIR/bin/SigDigger.no-strip
cd ..
```

BatchDrake, 2023-01-08:
> 1. Install the [FreeBSD VM image](https://download.freebsd.org/ftp/releases/VM-IMAGES/13.1-RELEASE/amd64/Latest/) into VirtualBox.
> 2. Boot FreeBSD
> 3. Install git: pkg install git
> 4. Check out the [...]

Hi Yuri,

I failed to find SigDigger in that git. Neither in `main` nor `2023Q1`. In what branch are you working?

yurivict, 2023-01-08:
SigDigger is in the main branch: https://cgit.freebsd.org/ports/tree/comms/sigdigger

```cd comms/sigdigger``` should get you there.

BatchDrake, 2023-01-09:
Hi Yuri,

Just found the directory, thanks! I also had time to look into your Makefile(s) more closely, and I now believe I understand what happened here.

According to the generated SuWidgets Makefile, you set the `PREFIX` to `/usr/local` explicitly, both in SigDigger and SuWidgets (which is fine). The problem is that, when you built SigDigger, you did not specify the `PREFIX` with which SuWidgets was installed. In your case, you should have run:

```
$ qmake PREFIX=/usr/local SuWidgetsLib.pro # In SuWidgets
$ qmake SUWIDGETS_PREFIX=/usr/local PREFIX=/usr/local SigDigger.pro # In SigDigger
```

Which is basically what @jeffpc did earlier in his example.

PS: I was going to direct you to the [Users' Manual](https://github.com/BatchDrake/SigDigger/raw/master/doc/SigDigger_User_Manual.pdf) (page 12). However, I now realize that I mentioned that everything was installed in `/usr/local` by default, which is not correct. In particular, SuWidgets installs its headers in Qt5 installation directory by default. Of course, this can be easily overriden by setting the corresponding `PREFIX`.

Hope this clarified all this a bit.

yurivict, 2023-01-09:
All these ports do - they call qmake/cmake with all default settings.
Your packages should behave reasonably with such default settings.

BatchDrake, 2023-01-09:
And they do: if you don't provide a `PREFIX`, they install in their default locations and find the required headers just fine. However, if you change they place in which SuWidgets is installed (which is what you did), you need to inform SigDigger's qmake about this change as well. Otherwise it is going to look for them in the default Qt5 installation directory, and fail.

In any case, I believe you have your fix now. Just pass `SUWIDGETS_PREFIX=/usr/local` to `qmake` in your Makefiles and tell me whether it works.

yurivict, 2023-01-09:
Ok, thanks, it works with ```SUWIDGETS_PREFIX=/usr/local```.

closed 2023-01-09 by yurivict