BatchDrake/SigDigger #200: DC Spike in middle of spectrum display.

John-Telek, 2023-02-22:
I get this spike in the middle of my display on the center frequency. The remove DC button does not do anything.
[image]

Any suggestions please ?

Cheers,
   John Telek

John-Telek, 2023-03-01:
This other spike is also present to the left of center. I'm getting this with no antenna. Any ideas ?

[image]

BatchDrake, 2023-03-01:
Does that spike move when you change the frequency?

John-Telek, 2023-03-01:
No it stays in the center. It's to do with the DC component. The DC remove button does nothing. I have no idea where the one on the left is from, and adjusting audio cutoff frequency can produce weird audio products.
Cheers,
      JT

BatchDrake, 2023-03-01:
Yeah, I would not expect the one in the center to move, I was curious about the one in the left. That can perfectly be an artifact of the RTL-SDR, caused by harmonics (if it moves with the spectrum) or images of something else (if it moves against the spectrum).

John-Telek, 2023-03-02:
Mate, would it be ok to post a video of my issues ?

Cheers,
   John

BatchDrake, 2023-03-02:
Sure, go ahead!

John-Telek, 2023-03-26:
I just tried the latest build and the DC spike is still there.

John.   :)

BatchDrake, 2023-03-26:
Fixed, launching a development build now.

John-Telek, 2023-03-26:
No worries mate. Thanks,

JT

BatchDrake, 2023-03-26:
https://github.com/BatchDrake/SigDigger/releases/tag/latest

John-Telek, 2023-03-26:
Still there.

JT

BatchDrake, 2023-03-26:
Press Ctrl+L and post a screenshot of the log window

BatchDrake, 2023-03-26:
I've just fixed something (DC correction worked for file sources only, not for devices without DC correction support), launching a new build now.

BatchDrake, 2023-03-26:
Download this again: https://github.com/BatchDrake/SigDigger/releases/tag/latest if this is not fixing it, this means that the SDR receiver supports DC offset correction but it is not working properly (on the device's side)

John-Telek, 2023-03-28:
Here is the Log window. Doesnt say anything.

JT

John-Telek, 2023-03-28:
Will doe mate. Thak you.  :)

JT

John-Telek, 2023-03-28:
The DC remove works on ather apps so my SDR dongle should be ok.

JT

John-Telek, 2023-03-28:
YAAYYY It works. :) :) :)

Thank you. How did you fix it pls ?

Cheers,
   JT.

BatchDrake, 2023-03-28:
I basically did it on software. I compute a sample average on every read and use it to update the DC offset, which is subtracted from the samples. 

I am still worried about the performance impact though, I will see whether it makes sense to use Volk to speed up things here.

closed 2023-04-03 by BatchDrake