BatchDrake/SigDigger #202: how to interface custom fpga data stream to SigDigger

hoarylea, 2023-04-03:
Mr. BatchDrake
I am a hardware engineer, focus on fpga development.
could you please guide me on how to interface fpga stream to SigDigger. thanks.

BatchDrake, 2023-04-03:
Hi,

SigDigger relies on SoapySDR for device interfacing, and uses it to read
I/Q samples from them in a sort-of agnostic way. It boils down to writing
the corresponding SoapySDR module, and make sure SigDigger is able to
recognize it (which should not be difficult).

This, or course, is in case you are intending to use your FPGA to deliver a
stream of I/Q samples to SigDigger. If you had something else in mind, we
may need to look for alternatives.

Cheers,

BatchDrake, 2023-04-12:
Hi @hoarylea 

Is there anything else I can help you with?

Cheers,