BatchDrake/SigDigger #204: Compilation breaking because of missing libpthread

John-Telek, 2023-04-20:
I had to new issues pop up all of a sudden when  using the build script. First it complained that it needed cmake 20.6 but only 18.3 was installed so I uninstalled the debian package and installed the latest version from cmake.org. Cool, one down. now these errors. I have all the required libs and dev files installed and libpthread is there but I think its a typo somewhere. The errors.....

[   OK   ] Running CMake (suscan)... 
[ FAILED ] Building suscan... 

--------------8<----------------------------------------
Try: make -j 12 -C suscan/build
CWD: /home/user/tmp/blsd-dir/build-root
/usr/bin/ld: cannot find -l-lpthread
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/suscan.dir/build.make:1459: libsuscan.so.0.3.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/suscan.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
--------------8<----------------------------------------

-l-lpthread  looks like a typo.

Any ideas mate ?

cheers,
   JT.

John-Telek, 2023-04-20:
Is there any way of seeing were it breaks because I have no idea where to start looking.
like make --trace or make -d  ?

Cheers,
  JT

BatchDrake, 2023-04-20:
We are aware, it is also breaking on GitHub for some reason but not locally in our computers. There is definitely a typo but we are struggling to reproduce. Can you run the script setting the env. var `BUILDTYPE=Debug` first? That would be extremely helpful!

John-Telek, 2023-04-20:
On Thu, 2023-04-20 at 02:15 -0700, Gonzalo José Carracedo Carballal
wrote:
> We are aware, it is also breaking on GitHub for some reason but not
> locally in our computers. There is definitely a typo but we are
> struggling to reproduce. Can you run the script setting the env. var
> BUILDTYPE=Debug first? That would be extremely helpful!

All right mate here is your first error.

[   OK   ] Creating builddirs... 
[   OK   ] Running CMake (sigutils)... 
[   OK   ] Building sigutils... 
[   OK   ] Deploying sigutils... 
[   OK   ] Testing deplyment of sigutils... 
[ NOTICE ] Sigutils headers:   -D_SU_SINGLE_PRECISION -DHAVE_VOLK -
I/home/user/tmp/blsd-dir/deploy-root/usr/include/sigutils -
I/home/user/tmp/blsd-dir/deploy-root/usr/include/sigutils/util
[ NOTICE ] Sigutils libraries: -L/home/user/tmp/blsd-dir/deploy-
root/usr/lib -lsigutils -lsndfile -lfftw3f -l-lpthread -lm -lvolk
[   OK   ] Running CMake (suscan)... 
[ FAILED ] Building suscan... 

--------------8<----------------------------------------
Try: make -j 12 -C suscan/build
CWD: /home/user/tmp/blsd-dir/build-root
/usr/bin/ld: cannot find -l-lpthread
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/suscan.dir/build.make:1459:
libsuscan.so.0.3.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/suscan.dir/all]
Error 2
make: *** [Makefile:136: all] Error 2
--------------8<----------------------------------------

Yell if you need anything else.

Cheers,
  JT.

John-Telek, 2023-04-20:
> This is some excellent result, thanks *a lot* for that. This narrows down
> the origin of the issue to sigutils, we will look into this ASAP

Mate, I am glad to be of help and while I’m definitely not a coder am I
kind of have an idea being able to build the software and sort most of the
shoes out myself I consider myself are a very experienced mechanic as
opposed to an engineer like yourself And it feels good to be part of
something communal. I’ve had a lot of activity with this group and a few
others mainly SDR plus plus and SDR angel and these three packages do it
for me at least anyway

Anyway, hope that’s helped, and I’m sure we’ll hear from you soon.
Cheers mate,
    John.

John-Telek, 2023-04-20:
https://github.com/John-Telek/Pi_SDR

This is my get hub page and yes, it’s basically another persons documentary
about pies and SDR in all sorts of other things but I’d like you to have a
look at the build instructions. I put up regarding sick digger and let me
know what you think please, if there’s any chance changes I should make or
if I have anything wrong More importantly

The sites meant to be more hands-on rather than package. And hopefully I’ve
documented enough so that pretty much anyone can do it.

Or meet you take care. Cheers
   John.

BatchDrake, 2023-04-20:
I've just checked it out, and I see it clones everything from master. That's fine, but maybe you want to pull the code from develop instead (as it includes the latest changes and has a bunch of cool features already). Give a look to the User's manual, chapter 2: https://github.com/BatchDrake/SigDigger/blob/master/doc/SigDigger_User_Manual.pdf

There are some other build methods that are slightly more automatized (like the blsd scripts) that you may prefer. I've also added a new script (`Scripts/build.sh`) that clones everything and builds SigDigger in a standalone directory, without needed to install anything system-wide.

antoniovazquezblanco, 2023-04-20:
Hi @John-Telek !

I suspect the build breaks due to some changes I made lately. Could you please tell me what version of Debian are you using so that I can replicate the issue?

Thanks for your help!

John-Telek, 2023-04-20:
I can do better. YUo cann clearly see the issue, I just dont know if I
can just modify it or does it get generated somewhere.

This was the giveaway.

[ NOTICE ] Sigutils libraries: -L/home/user/tmp/blsd-dir/deploy-
root/usr/lib -lsigutils -lsndfile -lfftw3f -l-lpthread -lm -lvolk
-------------------------------------------^^^^^^^^^^^-----------

***@***.***:~/tmp/blsd-dir/deploy-root/usr/lib$ more
pkgconfig/sigutils.pc

# sigutils library pkg-config file
# Automatically generated by CMAKE

prefix=/home/user/tmp/blsd-dir/deploy-root/usr
exec_prefix="${prefix}"
libdir="${prefix}/lib"
includedir="${prefix}/include"

Name: sigutils
Description: Digital signal processing utility library
URL: http://github.org/BatchDrake/sigutils
Version: 0.3.0 
Cflags: -I${includedir}/sigutils  -I${includedir}/sigutils/util -
D_SU_SINGLE_PRECISION -DHAVE_VOLK
Libs: -L${libdir} -lsigutils -lsndfile -lfftw3f -l-lpthread -lm -lvolk
***@***.***:~/tmp/blsd-dir/deploy-root/usr/lib$ 

Hope this helps. Cheers,
   John.

John-Telek, 2023-04-20:
More info for you.

I found the offending file,

 ~/tmp/blsd-dir/build-root/suscan/build$ grep pthread *

CMakeCache.txt:SIGUTILS_STATIC_LDFLAGS:INTERNAL=-L/home/user/tmp/blsd-
dir/deploy-root/usr/lib;-lsigutils;-lsndfile;-lfftw3f;-l-lpthread;-
lm;volk

edited the CMakeCache.txt abd got rid of the typo and ran make. 
it compiled but wouldnt go in the right spot so suscan wouldnt compile.

-l-lpthread is appearing everywhere.

Cheers,
  JT

BatchDrake, 2023-04-20:
It should be fixed now, try to build again from the beginning

John-Telek, 2023-04-20:
I just deleted everything and it all came good.
.

user@debian:~/tmp/SigDigger$ ./build.sh 
Welcome to...

    ____  _       ____  _                       
   / ___|(_) __ _|  _ \(_) __ _  __ _  ___ _ __ 
   \___ \| |/ _` | | | | |/ _` |/ _` |/ _ \ \'__|
    ___) | | (_| | |_| | | (_| | (_| |  __/ |   
   |____/|_|\__, |____/|_|\__, |\__, |\___|_|   ('s)   
            |___/         |___/ |___/           

...multi platform deployment script.

Attempting deployment on Linux (x86_64) with 12 threads
Date: Fri 21 Apr 2023 04:19:04 AEST
SigDigger release to be built: 0.3.0

[   OK   ] Locate Git... 
[   OK   ] Locate Make... 
[   OK   ] Ensuring pkgconfig availability...  
[   OK   ] Locating sndfile...  
[   OK   ] Locating libxml2... 
[   OK   ] Locating FFTW3... 
[   OK   ] Locating SoapySDR... 
[   OK   ] Locating Qt... 
[   OK   ] Locating CMake... 
[   OK   ] Cleaning deploy directory... 
[   OK   ] Cleaning buildroot... 
[   OK   ] Recreating directories... 
[   OK   ] Cloning sigutils (develop)... 
[   OK   ] Cloning suscan (develop)... 
[   OK   ] Cloning SuWidgets (develop)... 
[   OK   ] Cloning SigDigger (develop)... 
[   OK   ] Creating builddirs... 
[   OK   ] Running CMake (sigutils)... 
[   OK   ] Building sigutils... 
[   OK   ] Deploying sigutils... 
[   OK   ] Running CMake (suscan)... 
[   OK   ] Building suscan... 
[   OK   ] Deploying suscan... 
[   OK   ] Running QMake (SuWidgets)... 
[   OK   ] Building SuWidgets... 
[   OK   ] Deploying SuWidgets... 
[   OK   ] Running QMake (SigDigger)... 
[   OK   ] Building SigDigger... 
[   OK   ] Deploying SigDigger... 
[   OK   ] Removing unneeded development files... 
[   OK   ] Creating startup script for SigDigger 
[   OK   ] Creating startup script for suscli 
[   OK   ] Moving files out of /usr... 
[   OK   ] Remove empty /usr... 
[   OK   ] Setting permissions to wrapper scripts... 

Done. SigDigger compiled succesfully in /home/user/tmp/SigDigger/deploy-root
user@debian:~/tmp/SigDigger$ 

I just ran the build.sh script and it came good.

Cheers.

John-Telek, 2023-04-20:
All good thanks. What was the issue pls ?

JT

BatchDrake, 2023-04-20:
The issue was basically broken 3rd party CMake Modules. The CMake module for threads was prepending the compiler flag `-l` to the library name. While the rest of the CMakeLists.txt could work with that (silently ignoring it, ofc), the pkg-config file for sigutils expected that all library names were "as-is", with no prefixes whatsoever. And then, the result is what you saw here.

I basically coded a kludge to check for this buggy behavior, which seems to be fixed in the latest GNU/Linux distros.

John-Telek, 2023-04-20:
Awesome, thanks!

closed 2023-04-21 by BatchDrake