BatchDrake/SigDigger #207: Doppler tool

ea3hmj, 2023-05-08:
When I use the server to change the "Freq rate" value, it rounds to 2 decimal places.

BatchDrake, 2023-05-08:
What happens when you run `get` on the property, right after the set? Does the server reply with a rounded up value?

BatchDrake, 2023-05-09:
Hi again,

Just tested it myself. It seems to be saved properly:

```
set dopplertool:freq_rate 10.012345678
dopplertool:freq_rate = 10.012345678
get dopplertool:freq_rate
dopplertool:freq_rate = 10.012345678
```

[image]

The GUI is limited to 3 decimal places, and I believe that is fine for most manual corrections (1 mHz/s is something that varies slowly, in the order of several minutes to a few hours fonr 1 Hz change). On the other hand, I do acknowledge that it would be desirable to have a fixed precision output (right now it computes the display precision automatically). I will definitely change that.

BatchDrake, 2023-05-15:
Confirmed earlier and fixed