BatchDrake/SigDigger #212: Hack RF One.

John-Telek, 2023-05-17:
I cant seem to find support for the HRF1. I's it being considered ?

Would you consider a feature like Squelch triggered recording as a way to eliminate
disk wasteage and skip the blank bits.

JT.

BatchDrake, 2023-05-17:
Hi,

The AppImage should support it out of the box. If you are building from sources, the support comes indirectly from SoapySDR. This implies that you would need to have the SoapySDR module for HackRF installed.

On the other hand, the squelch-triggered recording is an interesting feature which I envisioned many times in the past and ended up forgetting. I've just opened a feature request so I don't forget this time: https://github.com/BatchDrake/SigDigger/issues/213

Cheers,

John-Telek, 2023-10-03:
I'd like to suggest adding a signal source feature for the hack rf as it would make a cool piece of gear. My  Dr SDR Facebook group shows where I make a specan and signal generator with gnuradio blocks. I see this app as a cool piece of open source hobbyist test gear using off the shelf bits.  I like :) A bit like Voyager.

closed 2023-11-07 by John-Telek