BatchDrake/SigDigger #217: AirSpy HF+ not visible

mashu, 2023-06-04:
Hi,

Prebuild AppImage crashes on dummy source with segmentation fault, so I build app from source. However no AirSpy HF+ is visiable on the list of sources.

BatchDrake, 2023-06-05:
Hi!

I've been checking out, and it seems that the AirSpy HF+ module is not included by default in my build machine, that's why the AppImages won't detect it. The dummy source should never crash nonetheless, it just the default when there's a missing device, and should fail gracefully.

Have you built the AirSpy HF+'s SoapySDR module from source too? Maybe that's why it is not detecting it. Please download it from here https://github.com/pothosware/SoapyAirspyHF

mashu, 2023-06-05:
Hi,

This is GNU/Linux Debian (sid) and I believe soapy airspy HF is already
installed
ii  soapysdr-module-airspy:amd64                       0.2.0-3+b1
                     amd64        Airspy device support for SoapySDR
(default version)
ii  soapysdr0.8-module-airspy:amd64                    0.2.0-3+b1
                     amd64        Airspy device support for SoapySDR

But after installing plugin you linked I can see it in sources. I just
don't know how to enable HF frequencies as I can't go below 9000 kHz

I was worried about this segfault so next I compiled SigDigger myself from
source, and it does not crash on a dummy source, but displays errors.

error: Failed to open RX stream on SDR device:

error: exception in "suscan_source_open_sdr(new)"
(build/CMakeFiles/suscan.dir/compiler_depend.ts:2513)

error: exception in "self->source = suscan_source_new(config)"
(build/CMakeFiles/suscan.dir/compiler_depend.ts:508)

error: Failed to initialize source

error: exception in "new->impl = (iface->ctor) (new, ap)"
(build/CMakeFiles/suscan.dir/compiler_depend.ts:606)

So at least it's not SIGSEGV.

BatchDrake, 2023-06-10:
Hi, just got my hands on an AirSpy HF+ myself and I can confirm it works. I will go ahead and close this issue. Feel free to reopen it again if something else comes up.

closed 2023-06-10 by BatchDrake

mashu, 2024-01-03:
As asked to re-open the issue.

The problem occurs again, but this time all modules are installed. I can tell that gqrx can receive with AirSpy HF but I don't see soapy selectable in SigDigger. I am using AppImage nightly builds.

Thanks!