BatchDrake/SigDigger #219: Error: "FrequencySpinBox.h: No such file or directory" during manual installation of SigDigger

OE3ALA, 2023-06-11:
Hello together,

I tried to install SigDigger according to the "Manual compilation" section of the [user manual](https://raw.githubusercontent.com/BatchDrake/SigDigger/master/doc/SigDigger_User_Manual.pdf). Unfortunately, I am getting the error message

```
/usr/lib/qt5/bin/uic ui/BookmarkManagerDialog.ui -o ui_BookmarkManagerDialog.h
g++ -c -pipe -D__FILENAME__=\"SigDigger\" -O2 -std=gnu++1y -Wall -Wextra -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DSIGDIGGER_HAVE_ALSA -D_SU_SINGLE_PRECISION -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -Iinclude -I/usr/include/x86_64-linux-gnu/qt5/SuWidgets -I/usr/local/include/suscan -I/usr/local/include/suscan/util -I/usr/local/include/sigutils -I/usr/local/include/sigutils/util -I/usr/local/include -I/usr/include/opus -I/usr/include/libxml2 -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o AppConfig.o App/AppConfig.cpp
In file included from include/PanoramicDialog.h:27,
                 from App/AppConfig.cpp:21:
./ui_PanoramicDialog.h:26:10: fatal error: FrequencySpinBox.h: No such file or directory
   26 | #include "FrequencySpinBox.h"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:2566: AppConfig.o] Error 1
```
after following commands:
```
$ cd SigDigger
$ qmake SigDigger.pro
$ make
```
Any help on fixing this issue? I guess it is related to QT5 5.15.3.

Many thanks in advance!

73de
Alex, OE3ALA

BatchDrake, 2023-06-11:
Hi,

It seems that you pulled SigDigger from develop but SuWidgets from master. You should pull all projects from develop.

You may find more convenient to use the blsd script (https://actinid.org/blsd) or the AppImage (releases -> development build, in this same page on the right)

OE3ALA, 2023-06-11:
Thanks for the fast answer!

I solved the issue ...

In the user manual the command `$ git clone https://github.com/BatchDrake/SuWidget` is wrong, it should be `$ git clone https://github.com/BatchDrake/SuWidgets`. Therefore, I downloaded the source code from master manually and installed it according to the instructions. This was the error ...

After deleting the SuWidgets folder and running the correct git clone command + reinstalling SuWidgets it worked out.

Closing the issue... :)

closed 2023-06-11 by OE3ALA