BatchDrake/SigDigger #220: Channel Inspection bandwidth for USB/LSB

srs4511351, 2023-06-16:
Computer:
Raspberry Pi 4
Debian GNU/Linux 12 (bookworm)
SDRplay RSP1A

develop branch

The Channel Inspection bandwidth spinner box value is twice the actual bandwidth on USB/LSB.
If Baseband audio preview is unchecked, the bandwidth indicator on the spectrum display doubles in width to match the Channel Inspection bandwidth value.

----Steve

BatchDrake, 2023-06-17:
Hi,

This is intended. The bandwidth indicator tells you the size of the channel you are going to open if you inspect it (either with AutoSquelch, Press to capture, or when you open an inspector). In this regard, SSB is treated as an AM signal with no PLL tracking and with a filtered lower side band (for USB) or a filtered upper side band (for LSB).

srs4511351, 2023-06-17:
This is difficult for the user who wants to listen to SSB.
Having to calculate every time is cumbersome.
Displaying a number that is different than what is seen on the display is counter-intuitive.

BatchDrake, 2023-06-17:
Well, this is an "audio preview". It is something you use to get an idea of
what the signal sounds like. The main use case is still analyzing unknown
(usually digital) signals.

Nonetheless, I do agree that maybe I can do something regarding the channel
representation. Maybe drawing a line in the limits of the other sideband?

Cheers,

srs4511351, 2023-06-18:
With the channel inspector open, I see the additional bandwidth display.
It is as wide as the bandwidth indicator.

This is a difficult problem, and your explanation makes sense. We need some way of explaining the situation to the SSB users. A dashed line different than the channel inspector, at the limit of the opposite sideband is a possibility.

BatchDrake, 2023-06-19:
I've just changed how the channels are represented when selecting SSB. Rebuild SuWidgets and SigDigger, and tell me if this works for you.

srs4511351, 2023-06-19:
On the spectrum display, the area being decoded is shaded in dray. That does tell the story.
On the waterfall, it isn't so clear, although the dashed line for the part that is not being decoded is a bit faded.
In either case, the entire bandwidth is indicated for the inspector.

I think it is OK.

closed 2023-06-19 by BatchDrake