BatchDrake/SigDigger #221: Channel Inspection bandwidth issues

srs4511351, 2023-06-16:
Computer:
Raspberry Pi 4
Debian GNU/Linux 12 (bookworm)
SDRplay RSP1A

develop branch

When changing Channel Inspection bandwidth with the dialogue box/spinner, it changes the bandwidth on the display, but does not take effect unless the frequency is changed or the bandwidth is changed by dragging it on the display.
On SSB, Channel Inspection bandwidth is seldom saved, defaulting to 33.333 kHz after restarting SigDigger. Rarely, it may save the Channel Inspection bandwidth.
On FM, when I set Channel Inspection bandwidth to 8.000 KHz. It defaulted to 7.074 kHz after restarting SigDigger.

----Steve

BatchDrake, 2023-06-17:
Reproduced, working on it.

BatchDrake, 2023-06-17:
It seems that the `bandwidthChanged` was not being emitted properly, and SigDigger was not being aware that you were changing the bandwidth from the Inspection panel (I usually do it with the mouse, adjusting the edges of the filterbox). It seems that it also prevented SigDigger from saving those changes. Pull & rebuild SigDigger and tell me if it works.

srs4511351, 2023-06-18:
After reinstalling SigDigger and modules, the bandwidth is having an immediate effect now.

It is also saving the selected bandwidth after restarting SigDigger.

closed 2023-06-18 by BatchDrake