BatchDrake/SigDigger #222: Strange behavior with the IFGR gain control

srs4511351, 2023-06-16:
Computer:
Raspberry Pi 4
Debian GNU/Linux 12 (bookworm)
SDRplay RSP1A

develop branch

IFGR initially uses an unknown default value when Run is pressed and doesn't reflect the Signal Source slider setting unless it is moved.
If you stop playback and start it again with the Run button, it goes back to the default value and moving the slider slightly restores the setting. RFGR seems to work correctly.
When pressing Run, `[WARNING] Not updating IFGR gain because AGC is enabled` is displayed in the terminal. Hardware AGC is not checked.
I think that either there should be a button to enable/disable AGC or IFGR shouldn't have any effect when AGC is enabled. In either case, AGC is apparent on the SSB signal wherever IFGR is set.
With the default value, the display only shows stronger signals, so I prefer to increase the value.

BatchDrake, 2023-06-17:
Hi,

It seems that, for whatever reason, SDRPlay starts up with the AGC on. This is very odd. I've just explicitly disabled AGC right before setting the initial gains to prevent this from happening. This change is in Suscan. Could you pull & rebuild Suscan and SigDigger (in this order) again?

Thanks,

srs4511351, 2023-06-18:
After reinstalling SigDigger and modules, the AGC still starts at the same setting when Run is pressed.
This setting does not seem to be related to the signals on the band, just a default value.
Selecting or deselecting Hardware AGC has no effect.

I do not believe that the AGC action that I hear on SSB is the hardware AGC, but  software feature. It is similar to the other SDR software that I have. They have AGC that works without hardware AGC, which does have an effect when selected.

BatchDrake, 2023-06-18:
Can you confirm you cloned the latest `suscan` from develop? I will keep looking into this further, but this is a behavior I cannot reproduce with other radios (like AirSpy).

On the other hand, what you hear in SSB is definitely the a software AGC. I've been a bit unhappy with it lately, because of what it sounds like when I reduce the bandwidth of the signal. Would you consider useful adding a control to disable it? Like a checkbox or something.

srs4511351, 2023-06-18:
I don't really know now to tell the version, but the suscan/build folder is dated 6/17/2023 23:20 EDT. The files inside have the same date.
The git checkout command says `Your branch is up to date with 'origin/develop'.`

I do not see this problem with my HackRF.

The software AGC is very useful with SSB and AM. I would not normally disable it.
It is very good except for the decay, which is too fast. If it could be set longer, it would work very nicely.
Sometimes applications will provide attack, decay and threshold settings. A checkbox to disable it could be added.

srs4511351, 2023-06-18:
Would it help to re-output the gain values every time Run is pressed?

srs4511351, 2023-06-19:
I rebuilt only suscan.
The gain levels are now saved and restored.
Actually, the previous IFGR from before the change was restored.

srs4511351, 2023-06-19:
Hardware AGC still has no effect.

BatchDrake, 2023-06-19:
Glad to hear that. It seems that the radio started itself with the AGC set to on by default (which is weird, but with SDRPlay everything is weird), and I had to disable it explicitly before setting the gains.

> Hardware AGC still has no effect.

I have just tested it again with AirSpy and it works perfectly. Also, it is strange, as I had to disable hardware AGC right before setting the gains because otherwise the SDRPlay API would complain with the warning you paster earlier.

My conclusion is that this one of the (many) SDRPlay issues. There's little I can do about it, except maybe opening a new issue in the SoapySDRPlay's github repo.

BatchDrake, 2023-06-19:
Just in case you want to compare it with my results, this is me setting and unsetting the hardware AGC in AirSpy, with no antenna.

[image]

BatchDrake, 2023-06-19:
Great, that is the exact behavior we are experiencing, so this makes it perfectly reproducible.

I am going to close this issue, is there anything else I could help you with?

Thanks,

srs4511351, 2023-06-19:
Except that SDRangel did work.

Are you considering adding options such as attack, decay and threshold settings for the software AGC?

BatchDrake, 2023-06-19:
> Except that SDRangel did work.

Yes, because it interfaces with the SDRPlay API directly. The problem is inside the SoapySDR module for SDRPlay.

> Are you considering adding options such as attack, decay and threshold settings for the software AGC?

I have several open fronts now, but feel free to open a new issue (as feature request) and I will update you from there. I prefer to keep issues on topic.

srs4511351, 2023-06-19:
No, I had the SoapySDR device selected in SDRangel.

BatchDrake, 2023-06-19:
I am afraid I cannot reproduce with any of the other radios. Also, just checked [SDRAngel's code](https://github.com/f4exb/sdrangel/blob/b4847f1d7a473d00f2be16ab227b01ebec8b5284/plugins/samplesource/soapysdrinput/soapysdrinput.cpp#L1152), and he adjusts the AGC the same way as I do.

The fact that the issue is easily reproducible by Gqrx makes me think that either SDRangel ignores the SoapySDR setting, or that it has been linked against a different version of the SDRPlay API, or something of the sort.

In either case, the SoapySDR API is what it is, and I am not able to maintain device-specific kludges for radios I do not have. But I've just opened an issue in the GitHub repo of the SDRPlay's module so you can follow up the discussion there: https://github.com/pothosware/SoapySDRPlay3/issues/73

srs4511351, 2023-06-19:
Thanks for opening the issue. gqrc is having the same trouble.

There is only one version of the SDRplay API on my system. The SDRangel device was the SoapySDR SDRplay driver.

srs4511351, 2023-06-30:
In https://github.com/pothosware/SoapySDRPlay3/issues/73 we discovered that the agc_setpoint default in SoapySDRplay3 is so low that strong signals cannot decrease IFGR further.
They added a agc_setpoint=-30 as a default. For SigDigger, you could choose to provide a setting for agc_setpoint or expect the users to update SoapySDRplay3.
I found that if I add agc_setpoint=-30 to the Soapy Device string, it would work.

----Steve