BatchDrake/SigDigger #225: Feature request: save bookmarks when added, modified or deleted

OH1KK, 2023-07-05:
SigDigger-0.3.0-x86_64-full-20230619-105310Z.AppImage

Currently ~/.suscan/config/bookmarks.yaml is saved when SigDigger is closed properly. This behavior is issue when SigDigger is running for days and I bookmark new signals as I found them. Then suddenly computers shuts down due power outage, all new bookmarks from past few days are gone.

To prevent this feature request: Save bookmarks when added, modified or deleted.

BatchDrake, 2023-07-05:
Got it, you are not the only one complaining and I am also getting tired of this behavior, I will change it soon.

BatchDrake, 2023-12-26:
Hi,

Since I had to write a small feature that performs config autosave every 30 minutes, I leveraged the existing code to implement this. Now config is dumped right after adding, changing or removing a bookmark. Give it a try.

Cheers,