BatchDrake/SigDigger #229: Enable Bias-T toggle?

hudsonvalleydigitalnetwork, 2023-09-10:
possible to add a toggle to turn off bias-t in SDR such as HackRF,  RTL-SDR V3/V4?   Using antenna with LNA built in make this handy and likely an easy add to your GUI.  Thanks for all your great work. I really enjoy this for SIGINT, especially the panoramic function with band start/stop.

BatchDrake, 2023-09-10:
Hi.

Go to settings, in Source Type select SoapySDR, and the device you want to use (your RTL-SDR in this case):

[image]

Then, click on _Device-specific tweaks_. A window like this would pop up:

[image]

Click _Add_, and a new row will be appended to the end of the list. Double click on the first column and write `setting:biastee`. Double click on the second column and write `true`. Save the settings and start the capture.

[image]

The Bias-T should be enabled now. You can verify if the underlying SoapySDR module has accepted this setting by opening the log window (View / Log, or just hit Ctrl+L) and reading the latest messages. In my case, it reports the following:

[image]

BatchDrake, 2023-09-10:
Okay, I have just realized you want to use this with the Panoramic Spectrum view, and I am afraid I have not exposed this feature in the Panoramic Spectrum window :( 

I am going to leave this open as a feature request, until I find time to write a Window to configure device-specific settings in a more general way.

PS: Have in mind that `biastee` is a RTL-SDR-specific setting. For HackRF and other receivers it may have different names, like `bias` or `bias-t`. Check the documentation (or go to the `Settings.cpp` file of the [corresponding device module source code in GitHub](https://github.com/pothosware/SoapyRTLSDR/blob/068aa77a4c938b239c9d80cd42c4ee7986458e8f/Settings.cpp#L628C31-L628C31) if there is none, as I did a few minutes ago) to make sure you are tweaking the right setting.