BatchDrake/SigDigger #236: Error building on Ubuntu 22.04

ToontjeM, 2024-02-07:
```
make -f Makefile.Release
make[1]: Entering directory '/opt/build/SuWidgets'
g++ -c -pipe -O2 -std=gnu++1y -Wall -Wextra -D_REENTRANT -fPIC -D_SU_SINGLE_PRECISION -DHAVE_VOLK -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I. -I/usr/local/include -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/Constellation.o Constellation.cpp
In file included from Constellation.h:28,
                 from Constellation.cpp:21:
Constellation.cpp: In member function ‘void Constellation::drawAxes()’:
Constellation.cpp:101:32: error: ‘I’ was not declared in this scope; did you mean ‘PI’?
  101 |     SUCOMPLEX delta = SU_C_EXP(I * angle);
      |                                ^
make[1]: *** [Makefile.Release:558: release/Constellation.o] Error 1
make[1]: Leaving directory '/opt/build/SuWidgets'
make: *** [Makefile:42: release] Error 2

```

closed 2024-02-07 by ToontjeM

sherdil6, 2024-02-21:
@ToontjeM Hi, I am facing this issue in SuWidgets. Tried to resolve this, however, nothing is working and I always end up in this error. Any idea how to resolve this please?

BatchDrake, 2024-02-21:
Are you building from master or develop?

sherdil6, 2024-02-21:
Used your script https://actinid.org/blsd by giving sudo rights.
Its resolved now.
Thank you