BatchDrake/SigDigger #240: fail to run appImage on WSL Ubuntu 22.04

hayguen, 2024-03-24:
running the latest AppImage from https://github.com/BatchDrake/SigDigger/releases,
it starts showing splash screen "generating FFT wisdom ..".
but then, i just get "Aborted" in the console.

BatchDrake, 2024-03-24:
Hi, thanks for reporting.

I cannot reproduce this. What are the specs of your system? Do you have any
SDR connected to your computer? Is "Aborted" the only text printed in the
terminal?

Cheers,

hayguen, 2024-03-24:
My system is Windows 11 .. with Windows Subsystem for Linux (WSL),
having  Ubuntu 22.04.4 LTS in WSL.
Unfortunately, no further text. Just "Aborted".

BatchDrake, 2024-03-24:
Wow, I'd be surprised if it worked under WSL. Anyways, WSL is still a hack,
and many things could fail. I bet it is something related to libusb being
unsupported under WSL (although it should work in WSL2). Perhaps it is
something Qt/X-Window related?

Nonetheless, I don't use Windows, so I'm afraid I cannot debug it myself :(
Still, if you have time, you may want to try to build it from sources. I
have this script (https://actinid.org/blsd) that downloads and builds the
sources from GitHub. If the problem persists, we can try to debug it step
by step from here.

Cheers,

hayguen, 2024-03-25:
Yes, unfortunately there's no "working"  libusb on WSL :-(
Thus, a compile build option for "don't try/build any SDR support" would be useful: having just file support.
will have a look at your build script.

Qt/Gui stuff works fine with the X-server of MobaXterm.

BatchDrake, 2024-03-25:
I used to have something for that during the major releases (the so-called "full" and "lite" versions), but it is too early to make a proper release and adjusting the build system for driverless development builds would take some time.

For the time being, I recommend we keep bulding from sources and/or use a real machine, or a VM (WSL does not count).

hayguen, 2024-03-25:
after installing a bunch of packages in the WSL-Ubuntu, i could build SigDigger successfully with the blsd script.
had to use `./blsd --disable-portaudio`.

starting SigDigger directly from the directory, which can be moved, i get the splash screen for a while.
then, SigDigger terminates and i have following on the console:
```
$ ./SigDigger
QOpenGLShaderProgram: could not create shader program
Vertex shader for simpleShaderProg (MainVertexShader & PositionOnlyVertexShader) failed to compile
QOpenGLShader: could not create shader
Fragment shader for simpleShaderProg (MainFragmentShader & ShockingPinkSrcFragmentShader) failed to compile
Errors linking simple shader:
QOpenGLShaderProgram: could not create shader program
Vertex shader for blitShaderProg (MainWithTexCoordsVertexShader & UntransformedPositionVertexShader) failed to compile
QOpenGLShader: could not create shader
Fragment shader for blitShaderProg (MainFragmentShader & ImageSrcFragmentShader) failed to compile
Errors linking blit shader:
QOpenGLShaderProgram: could not create shader program
Warning: "" failed to compile!
QOpenGLShaderProgram: could not create shader program
Warning: "" failed to compile!
QOpenGLShaderProgram: could not create shader program
Warning: "" failed to compile!
QOpenGLShaderProgram: could not create shader program
Warning: "" failed to compile!
The X11 connection broke: I/O error (code 1)
XIO:  fatal IO error 0 (Success) on X server "localhost:10.0"
      after 568 requests (568 known processed) with 0 events remaining.
ayguen@MSI:~/git/BatchDrake/blsd-dir/SigDigger$
```

looks, it tries to use 3D extensions .. and fails on them.
any chance to tell Qt or the X server, not to use any 3D stuff?
i would be happy with 2D.

hayguen, 2024-03-25:
not hearing audio. that might be an issue of WSL.
don't know, if i ever heard audio .. 
however, not that important for me

BatchDrake, 2024-03-25:
Hi,

I don't think it would work. It is not that I use 3D or anything, but I do
need OpenGL in order to run the shaders for smooth Waterfall zoom and
scrolling.

Anyways, this is an X server issue and I believe we've just hit a wall.
Perhaps switching the X server? Or using a real machine / VM?

Cheers,

BatchDrake, 2024-03-25:
There was a hacky way to setup PulseAudio, but I believe it is going to be
easier if you just install Ubuntu in a VM or a physical machine.

hayguen, 2024-03-25:
that works for me.
probably not perfectly smooth - but sufficient.
now, need to get familiar using SigDigger and the "inspectors" ..
and also, how to zoom at a specific frequency of the waterfall/spectrum ..
suppose, i need to Read The Fine Manual ;-)

BatchDrake, 2024-03-25:
Cool :D

Zoom is performed mostly with the mouse wheel. Mouse wheel in the
horizontal axis, horizontal zoom. Vertical axis, vertical zoom, etc. Also
you can drag the axes to adjust the zero point.

Cheers

hayguen, 2024-03-26:
Think, I should retest the App image on WSL - after having deactivated OpenGL stuff.