BatchDrake/SigDigger #241: file input configuration - way too complex

hayguen, 2024-03-24:
it could be as easy as opening a signal / wave file from File menu.
and additionally allow drag & drop such a file into the SigDigger window.

BatchDrake, 2024-03-25:
Hi,

Thanks for your feedback. This is not the first time somebody complains about this, and I have just decided that it is about time I get this fixed.

I have just written a feature so that you can load a capture file from the File menu, or by hitting Ctrl+O. I also added some minor fixes regarding UI performance. The affected projects are SuWidgets (develop branch) and SigDigger (also develop). If you are building with `blsd`, I'd just rerun it. If this is an AppImage, give me some minutes until it gets re-generated.

PS: Out of curiosity, is this running under WSL?

Thanks.

hayguen, 2024-03-25:
I'll test on WSL and also on "Real" Ubuntu machine ..
Drag & Drop would be really great: I usually open explorer or nautilus with several files .. and drag the files one by one into an analysis or audio app.

BatchDrake, 2024-03-25:
I have just implemented drag & drop real quick. Give it a try and tell me if it works.

hayguen, 2024-03-26:
Great! :-)
i'll test later the day.

hayguen, 2024-03-26:
works as it should on a native Ubuntu 22.04.4 LTS system (NOT WSL).
(noticed user manual states to `apt install libvolk1-dev`. but it has to be `libvolk2-dev`)
will test later on WSL

hayguen, 2024-03-26:
tested on WSL now:
unfortunately, Drag&Drop doesn't work,
neither with Windows-Explorer - nor with a `nautilus` file explorer install inside WSL Ubuntu Linux.

however, think, this is a WSL or MobaXterm problem.

closed 2024-03-26 by hayguen

BatchDrake, 2024-03-26:
Hi,

I would not expect it to work either. Drag & Drop is already a mess within
GNU/Linux systems and I wouldn't believe it if somebody dared to account
for all the corner cases needed to make it work between Windows and X apps
(like, how do you convert Windows paths into WSL paths transparently, etc)

Cheers,