BatchDrake/SigDigger #242: refer to blsd in the readme

hayguen, 2024-03-25:
in the https://github.com/BatchDrake/SigDigger/blob/master/README.md
there are instructions for building manually ..
but the blsd script is not referred.

BatchDrake, 2024-03-25:
Already present in the docs:
https://github.com/BatchDrake/SigDigger/blob/master/doc/SigDigger_User_Manual.pdf

hayguen, 2024-03-26:
Oh, there's a manual! I didn't know. How could i?
It's not referred: neither in the readme - nor on the website!
Please, put a link to it at both places.
But also in the readme on GitHub, with outdated build Infos.

BatchDrake, 2024-03-26:
You know, you are right. I am going to edit the master branch just to make people aware of the situation (and refer to the manual). Thanks!

BatchDrake, 2024-03-26:
Done, please check https://github.com/BatchDrake/SigDigger/blob/master/README.md

hayguen, 2024-03-26:
Haven't deeply read, but it looks much better now. Thank you!

Please link to the manual also on https://batchdrake.github.io/SigDigger/