BatchDrake/SigDigger #243: collect/document suitable tutorial videos per topic

hayguen, 2024-03-26:
found some videos on youtube. but didn't watch yet.

would think, that some would be a good amendment to the user manual.
would be great, if we could suggest/collect some videos per topic here!?

BatchDrake, 2024-03-26:
Truth is, I am not particularly happy with the manual. It was a first
attempt but I don't like the final style and I find it cumbersome to handle
at times.

A rewrite will take place at some point, and probably in a on-line form.
This would be a good place to link videos and stuff.

Cheers,

BatchDrake, 2024-03-26:
Speaking of which, do you know any good software/tool/framework to create
on-line documentation? :D Current docs are in LaTeX but I don't mind
adjusting the format by hand.

hayguen, 2024-03-26:
unfortunately, i'm also not an expert, when it comes to documentation.
i personally use dokuwiki for notes and similar stuff: https://codingspirit.de/dokuwiki/doku.php
when it comes to revisioned documentation, colleagues like .rst and sphinx, from which all kind of documents can be created .. see https://sphinx-tutorial.readthedocs.io/