BatchDrake/SigDigger #246: Many building errors

kenwawa01, 2024-06-17:
Hi,
Ive downloaded master version however when compiling (after earlier compilation of all required dependencies), I have the following errors (many) (I also downloaded suscan as master):

```
/usr/lib/qt5/bin/uic ui/BookmarkManagerDialog.ui -o ui_BookmarkManagerDialog.h
g++ -c -pipe -D__FILENAME__=\"SigDigger\" -O2 -std=gnu++1y -Wall -Wextra -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DSIGDIGGER_HAVE_ALSA -D_SU_SINGLE_PRECISION -DHAVE_VOLK -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -Iinclude -I/usr/include/x86_64-linux-gnu/qt5/SuWidgets -I/usr/local/include/suscan -I/usr/local/include/suscan/util -I/usr/local/include -I/usr/include/opus -I/usr/include/libxml2 -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o AppConfig.o App/AppConfig.cpp
In file included from include/AppUI.h:22,
                 from include/AppConfig.h:27,
                 from App/AppConfig.cpp:20:
include/Suscan/Analyzer.h:48:40: error: field ‘local_info’ has incomplete type ‘Suscan::suscan_analyzer_source_info’
   48 |     struct suscan_analyzer_source_info local_info;
      |                                        ^~~~~~~~~~
In file included from include/Suscan/Analyzer.h:37,
                 from include/AppUI.h:22,
                 from include/AppConfig.h:27,
                 from App/AppConfig.cpp:20:
include/Suscan/Messages/SourceInfoMessage.h:32:12: note: forward declaration of ‘struct Suscan::suscan_analyzer_source_info’
   32 |     struct suscan_analyzer_source_info *message = nullptr;
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from include/AppUI.h:22,
                 from include/AppConfig.h:27,
                 from App/AppConfig.cpp:20:
include/Suscan/Analyzer.h: In constructor ‘Suscan::AnalyzerSourceInfo::AnalyzerSourceInfo()’:
include/Suscan/Analyzer.h:53:7: error: ‘suscan_analyzer_source_info_init’ was not declared in this scope; did you mean ‘suscan_analyzer_source_info’?
   53 |       suscan_analyzer_source_info_init(&this->local_info);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       suscan_analyzer_source_info
include/Suscan/Analyzer.h: In destructor ‘Suscan::AnalyzerSourceInfo::~AnalyzerSourceInfo()’:
include/Suscan/Analyzer.h:59:9: error: ‘suscan_analyzer_source_info_finalize’ was not declared in this scope; did you mean ‘suscan_analyzer_source_info’?
   59 |         suscan_analyzer_source_info_finalize(&this->local_info);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         suscan_analyzer_source_info
include/Suscan/Analyzer.h: In constructor ‘Suscan::AnalyzerSourceInfo::AnalyzerSourceInfo(Suscan::suscan_analyzer_source_info*, bool)’:
include/Suscan/Analyzer.h:67:9: error: ‘suscan_analyzer_source_info_init’ was not declared in this scope; did you mean ‘suscan_analyzer_source_info’?
   67 |         suscan_analyzer_source_info_init(&this->local_info);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         suscan_analyzer_source_info
In file included from include/Suscan/Source.h:25,
                 from include/AppConfig.h:23,
                 from App/AppConfig.cpp:20:
include/Suscan/Analyzer.h:71:15: error: ‘suscan_analyzer_source_info_init_copy’ was not declared in this scope; did you mean ‘suscan_source_info_init_copy’?
   71 |               suscan_analyzer_source_info_init_copy(&this->local_info, ptr));
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
include/Suscan/Compat.h:30:11: note: in definition of macro ‘SU_ATTEMPT’
   30 |     if (!(expr)) {             \
      |           ^~~~
In file included from include/AppUI.h:22,
                 from include/AppConfig.h:27,
                 from App/AppConfig.cpp:20:
include/Suscan/Analyzer.h: In member function ‘Suscan::AnalyzerSourceInfo& Suscan::AnalyzerSourceInfo::operator=(const Suscan::AnalyzerSourceInfo&)’:
include/Suscan/Analyzer.h:100:9: error: ‘suscan_analyzer_source_info_finalize’ was not declared in this scope; did you mean ‘suscan_analyzer_source_info’?
  100 |         suscan_analyzer_source_info_finalize(&this->local_info);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         suscan_analyzer_source_info
In file included from include/Suscan/Source.h:25,
                 from include/AppConfig.h:23,
                 from App/AppConfig.cpp:20:
include/Suscan/Analyzer.h:103:13: error: ‘suscan_analyzer_source_info_init_copy’ was not declared in this scope; did you mean ‘suscan_source_info_init_copy’?
  103 |             suscan_analyzer_source_info_init_copy(
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
include/Suscan/Compat.h:30:11: note: in definition of macro ‘SU_ATTEMPT’
   30 |     if (!(expr)) {             \
      |           ^~~~
In file included from include/AppUI.h:22,
                 from include/AppConfig.h:27,
                 from App/AppConfig.cpp:20:
include/Suscan/Analyzer.h: In member function ‘uint64_t Suscan::AnalyzerSourceInfo::getPermissions() const’:
include/Suscan/Analyzer.h:115:26: error: invalid use of incomplete type ‘struct Suscan::suscan_analyzer_source_info’
  115 |       return this->c_info->permissions;
      |                          ^~
In file included from include/Suscan/Analyzer.h:37,
                 from include/AppUI.h:22,
                 from include/AppConfig.h:27,
                 from App/AppConfig.cpp:20:
include/Suscan/Messages/SourceInfoMessage.h:32:12: note: forward declaration of ‘struct Suscan::suscan_analyzer_source_info’
   32 |     struct suscan_analyzer_source_info *message = nullptr;
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from include/AppUI.h:22,
                 from include/AppConfig.h:27,
                 from App/AppConfig.cpp:20:
include/Suscan/Analyzer.h: In member function ‘uint64_t Suscan::AnalyzerSourceInfo::getSampleRate() const’:
include/Suscan/Analyzer.h:127:26: error: invalid use of incomplete type ‘struct Suscan::suscan_analyzer_source_info’
  127 |       return this->c_info->source_samp_rate;
      |                          ^~
In file included from include/Suscan/Analyzer.h:37,
                 from include/AppUI.h:22,
                 from include/AppConfig.h:27,
                 from App/AppConfig.cpp:20:
include/Suscan/Messages/SourceInfoMessage.h:32:12: note: forward declaration of ‘struct Suscan::suscan_analyzer_source_info’
   32 |     struct suscan_analyzer_source_info *message = nullptr;
```

and many more but I hope its sufficient?

BatchDrake, 2024-06-17:
Do not build from master. Neither SigDigger nor SuWidgets, build from
develop instead. See
https://github.com/BatchDrake/SigDigger/blob/master/README.md#notes-for-first-users

kenwawa01, 2024-06-17:
Ok solved. Thank you. Although for me it was very confusing as basing on ur description I was expected that when wanting to have stable version, I should get master version and this is even no compiling...

BatchDrake, 2024-06-17:
We are in the middle of the process of merging develop to master, but that takes time and cannot be done at once. But duly noted, and I agree it is confusing, we need to finish this merge ASAP