BatchDrake/SigDigger #247: Exception in "suscan_source_device_get_info(self->device, self->channel, &info)"

kenwawa01, 2024-06-17:
Hi!
When using this software, Im getting the following exception - not sure what is causing this but I hope it might be helpful:

exception in "suscan_source_device_get_info(self->device, self->channel, &info)" (analyzer/source/impl/soapysdr.c:612)
analyzer/source/impl/soapysdr.c
suscan_source_soapysdr_get_freq_limits
line 612

BatchDrake, 2024-06-17:
What device have you chosen for capture?

kenwawa01, 2024-06-18:
https://pl.aliexpress.com/item/1005007039112063.html

BatchDrake, 2024-06-19:
How have you installed the RTL SoapySDR module/libhackrf? If you built everything from sources, chances are that you forgot to update the system.d rules, or that the kernel module is taking ownership on the device.

Does it work with other SDR applications, like Gqrx?

kenwawa01, 2024-06-19:
Concerning the first question, difficult to answer now as I decided to uninstall.

Yes, Gqrx works.

BatchDrake, 2024-06-19:
Okay, so in that case I am going to close this issue as cannot reproduce. Feel free to reopen it if you reinstall the software at some point in the future and this error keeps happening.

closed 2024-06-19 by BatchDrake