BatchDrake/SigDigger #252: [UI] waterfall is not in sync with mouse

jedi7, 2024-08-04:
Hi,

the waterfall frequency change by dragging is not in sync with the mouse cursor.
Steps to reproduce:
* click left on the horizontal row with frequencies (for example 400MHz)
* start dragging to left so the freq 400MHz will move to the left.
* stop mouse movement
PROBLEM: the mouse cursor does not points to 400MHz anymore
EXPECTED: the mouse cursor still points to 400MHz

When I try the same at version 0.3.0 then the behavior is correct.

BatchDrake, 2024-08-04:
Hi,

I will test it ASAP today. In the meantime, can you confirm you are
building Suscan and Sigutils from master, and SuWidgets and SigDigger from
develop?

Cheers,

jedi7, 2024-08-04:
I apologize I rebuilded all with latest versions and now it is again correct. So closing issue .

closed 2024-08-04 by jedi7