BatchDrake/SigDigger #253: Frequency digit at GHZ scale are trimmed on hackrf, so I got 400MGZ instead of 2400MGZ on device

WhiteWind4, 2024-08-21:
I am successfully working with qspectrumanalyzer and other tools for hackrf, but SigDigger was showing me nothing. After some investigation I found that it just drops GHZ digit, so when I am receiving at 5.6Ghz it showing 600 000 000 on my hackrf  portapack screen. 
I am on Ubuntu 24.04
SigDigger 0.3.0 appimage

BatchDrake, 2024-08-21:
Hi!

Okay, that's weird. I recall capturing 5.8 GHz stuff with my HackRF (particularly, NTSC signal from a FPoV camera) using SigDigger. Perhaps it is an issue in the SoapySDR module?

I am currently at work and cannot test this right now, but I will try later today at home. In the meantime, I have just triggered a new development build just to rule out any problems with the underlying module. I will get back to you once it is finished.

BatchDrake, 2024-08-21:
Try downloading this version: https://github.com/BatchDrake/SigDigger/releases/tag/latest