BatchDrake/SigDigger #259: suscli with AppImage

paccerdk, 2024-10-04:
I'm trying to get remote connections working, using the AppImage, 

I was browsing the source code, and https://github.com/BatchDrake/SigDigger/issues/154 seems to hint, that running "./SigDigger-0.3.0-x86_64-full.AppImage -t suscli" might be a way of calling suscli when using the AppImage?

calling with "-t RMSViewer" works fine

The fact that [SUSCLI_ARGS](https://github.com/BatchDrake/SigDigger/blob/1c2613e22a5d527bee11f817f3e51109a415a21e/Scripts/AppRun#L37) is empty makes me think that i might be wrong about my conclusion, but i'm not sure

Is it currently not possible to use suscli and/or run a remote server with the current AppImage?
If not, maybe it might be worth considering

Additionally, a --server parameter could possibly also be a neat shortcut to have

BatchDrake, 2024-10-05:
Rename the AppImage file to just `suscli`. Or even better, create a soft link to it named suscli, like:

```
% ln -s /SigDigger-0.3.0-x86_64-full.AppImage suscli
% ./suscli
```

The `-t` option is only for the SigDigger executable, to choose which graphical tool to run. If you rename the AppImage, it will look for the appropriate executable inside the bundle.

paccerdk, 2024-10-05:
Oh, I see. The symbolic link approach is pretty neat, thanks!
Should i keep this ticket open for the suscli with -t option / adding the --server suggestions?

Its mostly unimportant QoL, feel free to close the ticket 
Thanks again!

BatchDrake, 2024-10-05:
Nah, it's a good a idea to have a GUI to quickly start up a server (that's how it would fit under the -t option). I am going to keep this open as a feature request. Thanks!